meshtastic/firmware #6146: [Feature Request]: Expose LFS Bad Block Errors to Firmware

Xaositek, 2025-02-24:
### Platform

NRF52

### Description

The intent of this is to explore the option of exposing the LFS Bad Block errors at a firmware level which in turn can be used for user experience.

We know that after 2.5.20, the only way to clear bad blocks is to do a wipe/reload, but issuing a blanket "everyone should do this" is a bit too board. Helping users understand if they should perform these actions would be beneficial and it starts at the firmware level.

garthvh, 2025-04-18:
Why would end users need this info?

Xaositek, 2025-04-18:
It's likely not directly useful to the end user, but in the user experience section, it is useful as warnings could be generated in the iOS or Android apps to ensure can take necessary actions to correct this. Similar to Windows or macOS prompting for disk checks.