meshtastic/firmware #6156: [Bug]: Unit consistently reboots on wake, after going to Light Sleep using TIO on MacOS terminal (USB C)

1nerdherder, 2025-02-25:
### Category

Serial

### Hardware

T-Deck

### Firmware Version

2.5.23.bf958ed Alpha

### Description

How to reproduce (see file attached):

(1) Connect tdeck to Mac using USBC to USB A cable via a USB C to A adapter.
(2) open Terminal
(3) find serial port by running ls /dev
(4) connect TIO to appropriate serial port
(5) watch serial trace go by and wait for:
INFO  | ??:??:?? 711 [PowerFSM] lsEnter begin, ls_secs=300  
INFO  | ??:??:?? 711 [PowerFSM] setup LORA_DIO1 (GPIO45) with wakeup by gpio interrupt
Then the serial stream stops.
(6) Give the trackball a normal short downward press - there will be no response
(7) Wait 40-60 seconds
(6) you'll see the reboot screen and the reboot trace in the serial stream.

I can repeat this on multiple Tdecks (plus and old) running multiple different versions of firmware.

Thanks and let me know if you need anything else.

### Relevant log output

```Shell
DEBUG | ??:??:?? 652 [Router] Module 'routing' considered
DEBUG | ??:??:?? 653 [RadioIf] Started Tx (id=0xc1da527b fr=0xea069358 to=0x75e1c1bc, WantAck=0, HopLim=3 Ch=0x8 encrypted len=104 hopStart=3 priority=10)
DEBUG | ??:??:?? 653 [RadioIf] Packet TX: 1009ms
DEBUG | ??:??:?? 653 [RadioIf] 0 packets remain in the TX queue
DEBUG | ??:??:?? 654 [RadioIf] Completed sending (id=0xc1da527b fr=0xea069358 to=0x75e1c1bc, WantAck=0, HopLim=3 Ch=0x8 encrypted len=104 hopStart=3 priority=10)
DEBUG | ??:??:?? 659 [RadioIf] Lora RX (id=0xc1da527b fr=0xea069358 to=0x75e1c1bc, WantAck=0, HopLim=2 Ch=0x8 encrypted len=104 rxSNR=6 rxRSSI=-65 hopStart=3)
DEBUG | ??:??:?? 659 [RadioIf] Packet RX: 1009ms
DEBUG | ??:??:?? 659 [Router] Rx someone rebroadcasting for us (id=0xc1da527b fr=0xea069358 to=0x75e1c1bc, WantAck=0, HopLim=2 Ch=0x8 encrypted len=104 rxSNR=6 rxRSSI=-65 hopStart=3)
DEBUG | ??:??:?? 659 [Router] Didn't find pending packet
DEBUG | ??:??:?? 659 [Router] Found existing packet record for fr=0xea069358,to=0x75e1c1bc,id=0xc1da527b
DEBUG | ??:??:?? 659 [Router] Add packet record (id=0xc1da527b fr=0xea069358 to=0x75e1c1bc, WantAck=0, HopLim=2 Ch=0x8 encrypted len=104 rxSNR=6 rxRSSI=-65 hopStart=3)
DEBUG | ??:??:?? 659 [Router] Ignore dupe incoming msg (id=0xc1da527b fr=0xea069358 to=0x75e1c1bc, WantAck=0, HopLim=2 Ch=0x8 encrypted len=104 rxSNR=6 rxRSSI=-65 hopStart=3)
DEBUG | ??:??:?? 659 [Router] cancelSending id=0xc1da527b, removed=0
DEBUG | ??:??:?? 659 [Router] Incoming msg was filtered from 0xea069358
DEBUG | ??:??:?? 661 [Power] Battery: usbPower=1, isCharging=1, batMv=4999, batPct=100
INFO  | ??:??:?? 681 [PowerFSM] Turn off screen
DEBUG | ??:??:?? 681 [Power] Battery: usbPower=1, isCharging=1, batMv=5022, batPct=100
DEBUG | ??:??:?? 701 [Power] Battery: usbPower=1, isCharging=1, batMv=5031, batPct=100
INFO  | ??:??:?? 711 [PowerFSM] lsEnter begin, ls_secs=300
INFO  | ??:??:?? 711 [PowerFSM] setup LORA_DIO1 (GPIO45) with wakeup by gpio interrupt

[13:20:45.852] Disconnected
[13:20:46.874] Connected to /dev/tty.usbmodem8CBFEA0693581
??@INFO  | ??:??:?? 1 

//\ E S H T /\ S T / C

INFO  | ??:??:?? 1 Booted, wake cause 0 (boot count 1), reset_reason=reset
DEBUG | ??:??:?? 1 Filesystem files (65536/3538944 Bytes):
DEBUG | ??:??:?? 1  /messages/log_000001.log (3990 Bytes)
DEBUG | ??:??:?? 1  /messages/log_000002.log (3988 Bytes)
DEBUG | ??:??:?? 1  /messages/log_000003.log (437 Bytes)
DEBUG | ??:??:?? 1  /prefs/channels.proto (92 Bytes)
DEBUG | ??:??:?? 1  /prefs/config.proto (195 Bytes)
DEBUG | ??:??:?? 1  /prefs/db.proto (1625 Bytes)
[... 202 lines cut ...]
DEBUG | ??:??:?? 20 [GPS] Probe for GPS at 38400
DEBUG | ??:??:?? 20 [GPS] Set Baud to 38400
DEBUG | ??:??:?? 20 [GPS] Trying $PDTINFO (UC6580)...
DEBUG | ??:??:?? 21 [GPS] Trying $PDTINFO (UM600)...
DEBUG | ??:??:?? 22 [GPS] Trying $PCAS06,1*1A (ATGM336H)...
DEBUG | ??:??:?? 22 [GPS] Trying $PCAS06,1*1A (ATGM332D)...
DEBUG | ??:??:?? 23 [GPS] Trying $PAIR021*39 (AG3335)...
DEBUG | ??:??:?? 23 [GPS] Trying $PAIR021*39 (AG3352)...
DEBUG | ??:??:?? 24 [GPS] Trying $PQTMVERNO*58 (LC86)...
DEBUG | ??:??:?? 24 [GPS] Trying $PCAS06,0*1B (L76K)...
DEBUG | ??:??:?? 25 [GPS] Trying $PMTK605*31 (L76B)...
DEBUG | ??:??:?? 25 [GPS] Trying $PMTK605*31 (PA1616S)...
DEBUG | ??:??:?? 26 [GPS] Trying $PMTK605*31 (LS20031)...
DEBUG | ??:??:?? 26 [GPS] Module Info : 
DEBUG | ??:??:?? 26 [GPS] Soft version: ROM SPG 5.10 (7b202e)
DEBUG | ??:??:?? 26 [GPS] Hard version: 000A0000
DEBUG | ??:??:?? 26 [GPS] Extensions:5
DEBUG | ??:??:?? 26 [GPS]   FWVER=SPG 5.10
DEBUG | ??:??:?? 26 [GPS]   PROTVER=34.10
DEBUG | ??:??:?? 26 [GPS]   MOD=MIA-M10Q
DEBUG | ??:??:?? 26 [GPS]   GPS;GLO;GAL;BDS
DEBUG | ??:??:?? 26 [GPS]   SBAS;QZSS
DEBUG | ??:??:?? 26 [GPS] Protocol Version:34.10
DEBUG | ??:??:?? 26 [GPS] ProtVer=34
```

mverch67, 2025-02-25:
This affects (probably) only serial connections to a Mac computer, as I cannot reproduce it with a PC, but only with my (old) MacBook. So I think it may be an electrical issue (hardware) and not necessarily software related.

1nerdherder, 2025-02-25:
Here is something interesting I just tested:
After the unit goes to light sleep with TIO running, and waiting for the serial stream to stop, then pressing the trackball and preparing to wait for the 40-60 seconds for the reboot - during this time I suspended TIO (CTRL-T, CTRL-Q) and the screen awoke and didn't reboot.  I checked the uptime and it persisted from the previous run.  So the serial port is hanging something in the wake up sequence.  Turning TIO off - seems to release it.

Xaositek, 2025-02-26:
I had this happen, I mentioned it to @thebentern. I was able to reproduce this but only with the Apple USB-C to USB-C cable, I swapped over to a different off-brand cable and this behavior stopped.

mverch67, 2025-02-26:
> I had this happen, I mentioned it to [@thebentern](https://github.com/thebentern). I was able to reproduce this but only with the Apple USB-C to USB-C cable, I swapped over to a different off-brand cable and this behavior stopped.
 
Is the mentioned off-brand cable a data cable or just a power cable?

Xaositek, 2025-02-26:
> Is the mentioned off-brand cable a data cable or just a power cable?

I have now retested with three separate cables (including the one I thought was okay) - Apple USB-C to USB-C, etguuds USB-C to USB-C, and QuanLink USB-C to USB-C.

What I identified is that once you see the lines:
INFO  | ??:??:?? 75 [PowerFSM] lsEnter begin, ls_secs=300
INFO  | ??:??:?? 75 [PowerFSM] setup LORA_DIO1 (GPIO45) with wakeup by gpio interrupt

Somewhere between 30-60 seconds later (if no packets are received), the device will go unresponsive and clicking the trackpad will not wake the device. A reboot will occur approximately 30 seconds after that.

If you wake up the device before it goes unresponsive, it will continue to function as expected.

Two factors seem to be limiting the impact of this if any of these USB cables are plugged into my Studio Display AND Belkin Connect USB-C to 4-Port USB-C Hub (https://www.amazon.com/dp/B0BY27XSJ3) I can not reproduce this, they MUST be plugged directly into an M3 MacBook Pro, for the purposes of the testing the USB-C port on the right side was used.

1nerdherder, 2025-02-26:
I have used different cables and seen the same result (all from the right side USB-C port.  I even have gone from a USB-C to USB-A adapter, then through a USB-A to USB-C cable to the t-deck (sometimes through a USB-C to USB-C power monitor) - still have same wake up issue.

closed 2025-04-20 by github-actions[bot]