meshtastic/firmware #6166: [Bug]: T-Deck w/ 2.6 MUI, Maps does not show fixed position when GPS is unavailable

antijingoist, 2025-02-27:
### Category

Other

### Hardware

T-Deck

### Firmware Version

2.6.0.0b106d4

### Description

On T-Decks without a GPS, GPS set to NOT_PRESENT, and using fixed_position to manually set location:

Opening Maps defaults to Big Ben in UK, and assumes device position is at that location. 

Device position is still sent out accurately, this only appears to affect the maps.

Expected: 
Location set using fixed position would be what Maps shows.

### Relevant log output

```Shell

```

caveman99, 2025-02-27:
Which map do you mean? on the actual t-deck or in the andriod/ios app?

antijingoist, 2025-02-27:
The map on the T-Deck

mverch67, 2025-02-27:
The field _fixed position_ is not / cannot be used for the MUI map. Instead the map has it's own persistent home position which can be set by long pressing the crosshair button and returned to by short pressing.

antijingoist, 2025-02-27:
If fixed_position coordinates are write-only, keeping a separate  home location makes sense. I really did try to look into it, but didn't find anything with that information. 😁 

Just to clarify:

* fixed_position coordinates cannot be read using the API MUI uses. Is this something that could change in the future or doubtful?
* the change you are offering, is this to set the fixed_position field to the home position when you long-press the crosshairs? ie: "No GPS detected, set your position" type of thing? Usability wise, it seems like a great idea.