meshtastic/firmware #6167: [Bug]: T3S3 (ESP32S3) Wont flash using ESPTool or PIO Direct from CLI on Linux (rPi Bookworm)

Talie5in, 2025-02-27:
### Category

Other

### Hardware

T-Beam S3

### Firmware Version

2.5.any

### Description

Ran into an issue where I have not been able to use ESPTool or PIO to Flash a T3S3 without powering on while holding the BOOT button to put into JTAG mode.

For a remote node I have connected to a rPi this was a bit annoying having to manually intervene to do updates.

I have however been able to get around this by modifying the platform.io for the t3s3 variant to include the below options in the build_flags.

  -DARDUINO_USB_MODE=1
  -DARDUINO_USB_CDC_ON_BOOT=1

This puts the unit into JTAG mode on boot (similar to a Heltec v3)

Not sure if both are needed, but once I have done a ./device-update with the build image (with a manual process first), I can now flash firmware without having to manual power cycle and hold the boot button.

[image]

Sourced from : https://github.com/espressif/arduino-esp32/issues/6762#issuecomment-1182821492

### Relevant log output

```Shell

```

mverch67, 2025-03-02:
In principle all esp32-s3 should have both of these values set to 1 as it gives a lot of advantages (e.g. esp logs are visible, flash without the need to manually go into DFU mode). Previously, when the first LoRa devices with esp32-s3 were integrated into meshtastic it was not allowed to set both values to 1. In the meantime this has been changed by espressif and could be adopted.

Only remaining issue is that all MUI devices which have another concurrent thread would mess up the debug log with intermangled log messages when the above settings are applied.

Talie5in, 2025-03-02:
> Only remaining issue is that all MUI devices which have another concurrent thread would mess up the debug log with intermangled log messages when the above settings are applied.

Would a simple fix be to make the MUI logs output I'm the same "format" as the current logs?

mverch67, 2025-03-10:
> In this case I believe it outputs the system boot logs via the Serial pin interface.

Yes exactly. But it's not only the esp logs that are messed up but also all the LOG_DEBUG from device-ui lib and firmware which is not the case if it's set to 0.

closed 2025-05-02 by github-actions[bot]