meshtastic/firmware #6170: Remove unnecessary null pointer checks

elfring, 2025-02-27:
[An extra null pointer check is not needed in functions](https://isocpp.org/wiki/faq/freestore-mgmt#delete-handles-null "Do I need to check for null before delete p?") like the following.
- [AccelerometerThread::clean](https://github.com/meshtastic/firmware/blob/b46bf1638552a28b3d9a0d8250330d7cf5bae584/src/motion/AccelerometerThread.h#L163-L166 "AccelerometerThread::clean function")
- [AudioThread::stop](https://github.com/meshtastic/firmware/blob/b46bf1638552a28b3d9a0d8250330d7cf5bae584/src/AudioThread.h#L44-L47 "AudioThread::stop function")
- [CryptoEngine::aesSetKey](https://github.com/meshtastic/firmware/blob/b46bf1638552a28b3d9a0d8250330d7cf5bae584/src/mesh/CryptoEngine.cpp#L164-L167 "CryptoEngine::aesSetKey function")

thebentern, 2025-02-27:
@elfring we welcome your pull-requests for these code quality improvements if you're willing to contribute to the project. :-)

elfring, 2025-02-27:
:thought_balloon: Would you become interested to [use a development tool like “clang-tidy” for corresponding source code adjustments](https://clang.llvm.org/extra/clang-tidy/checks/readability/delete-null-pointer.html "Deletion of unnecessary null pointer checks")?

thebentern, 2025-02-27:
> 💭 Would you become interested to [use a development tool like “clang-tidy” for corresponding source code adjustments](https://clang.llvm.org/extra/clang-tidy/checks/readability/delete-null-pointer.html)?

Possibly so. I'm curious if we can integrate this into our existing trunk.io configuration, since it already wrangles clang-format as a linter. @vidplace7

closed 2025-03-21 by thebentern

elfring, 2025-03-21:
:eyes: Was another update candidate left over so far?
https://github.com/meshtastic/firmware/blob/0951fdd49b67c00c3c075b4236b12af7fb04d507/src/mesh/CryptoEngine.cpp#L224-L229