meshtastic/firmware #6182: [Feature Request]: Add dynamic location precision. Show my precise location, unless I'm at home.

Woutvstk, 2025-02-28:
### Platform

Cross-Platform

### Description

Just an idea, not necessarily a feature request: I personally like seeing other Meshtastic users on the map in the app. I don't, however, want to forget to turn off my location sharing and then tell others/mqtt where I live.

How about an option to reduce location precision when I come within X km of point Y (my house)?

b8b8, 2025-02-28:
A "home zone" is an excellent idea to add. Also the option to completely and automatically disable location when within a certain distance from home then re-enable when leaving that zone.

RCGV1, 2025-02-28:
I can already implement this on iOS using shortcuts. I can add a shortcut to fuzz your location but I cannot add a precise location shortcut because of the privacy policy.

Woutvstk, 2025-02-28:
Yes, but if you give people the guarantee, it should be implemented in firmware. Devices that use their built-in gps should also turn off/down their location automatically.

RCGV1, 2025-02-28:
I do not think that is something that should be added to your firmware. As you would have to enter your home location into the device, so if your esp32 gets captured your home location can be retrieved. This should be done in client instead.

ianmcorvidae, 2025-02-28:
A client implementation doesn't work at all for anything that's not connected to a client (such as a vehicle tracker), and it'd be delayed (unacceptably, I suspect) for anything with a built-in GPS even if a client's connected. Part of the point of using position precision regions for this would be to ensure the location stored on the device is itself imprecise -- you'd only store the center of the position precision quasi-square plus the precision level, so you wouldn't reveal more precise information than what could be deduced from broadcasts anyway.

I don't think it's bad to have multiple ways to do this, so client implementations also make sense, but I think there's good reasons to make it available at the firmware level.

garthvh, 2025-02-28:
Currently not possible to send imprecise locations from the client at all https://github.com/meshtastic/firmware/pull/3752

Woutvstk, 2025-02-28:
> There's been talk in the past about doing some sort of a geofencing module for this sort of a case. My idea at the time was to set a location (or maybe several, depending what we can do) + a position precision, and allow settings to be changed for the "inside the zone" case. So for this case, you'd set precise location normally, but set an appropriate zone to turn off position (or possibly just to set position to the same position precision as the zone itself).
> 
> I'd still like to see it happen, unsure how much work needs to be done to make it happen. But +1 from me, anyway.

I like the idea of "Within this zone, just show that I'm within the zone." You could easily set a zone in the centre of your town/city and the zone size as large as the city. It would simplify the settings, and just seeing "5 people within this town" makes for a clean map.

> I do not think that is something that should be added to the firmware. As you would have to enter your home location into the device, so if your esp32 gets captured your home location can be retrieved. This should be done in client instead.

The centre of the home zone doesn't have to be on your home location

> Currently not possible to send imprecise locations from the client at all [#3752](https://github.com/meshtastic/firmware/pull/3752)

So this would only be possible to implement for the location determined by the node's onboard gps.

b8b8, 2025-03-01:
> I do not think that is something that should be added to the firmware. As you would have to enter your home location into the device, so if your esp32 gets captured your home location can be retrieved. This should be done in client instead.

It has been mentioned before on the discord that physical node security is up to the user. if your node is captured you're SOL. This needs to be a firmware change if its implemented. 

Your "home position" can be a 2 blocks away with a 3km radius or general advice could be given to use the closest park or postal office as your home. This is an easily surmountable problem.

garthvh, 2025-03-01:
> > I do not think that is something that should be added to the firmware. As you would have to enter your home location into the device, so if your esp32 gets captured your home location can be retrieved. This should be done in client instead.
> 
> It has been mentioned before on the discord that physical node security is up to the user. if your node is captured you're SOL. This needs to be a firmware change if its implemented.
> 
> Your "home position" can be a 2 blocks away with a 3km radius or general advice could be given to use the closest park or postal office as your home. This is an easily surmountable problem.
> 
> Maybe the first implementation for simplicity's sake would just be to turn off location sharing within 3km of set home location.

While "home" can feel like a bigger  leak than other locations the legal frameworks don't really make this distinction, and in many cases locations outside this home zone are equally ore more sensitive.

Woutvstk, 2025-03-01:
> While "home" can feel like a bigger leak than other locations the legal frameworks don't really make this distinction, and in many cases locations outside this home zone are equally or more sensitive.

So, not using the name "Home". How about just calling it "Location sharing geofencing" and make it possible to add a single point and radius where location data is not shared?

fifieldt, 2025-03-01:
Strava calls them "privacy zones". For reference, here are the options to choose from: 

[image]

garthvh, 2025-03-01:
> Sports app Strava calls them "privacy zones". For reference, here are the options to choose from:

This could work with some additional features in place, the strava functionality depends on user accounts and encryption being in place for the location data.

garthvh, 2025-03-01:
> So, not using the name "Home". How about just calling it "Location sharing geofencing" and make it possible to add a single point and radius where location data is not shared?

Previously we had discussed having a fixed sized array with a few different points and probably a radius setting.  We are having a risk assessment done on the location issues generally and once those reccomendations are in place we can probably re-visit this geofencing type functionality within a framework that better respects user privacy.

Woutvstk, 2025-03-01:
> Previously we had discussed having a fixed sized array with a few different points and probably a radius setting. We are having a risk assessment done on the location issues generally and once those [...]

I might be oversimplifying it, but isn't it relatively easy to add a condition just before a location data packet is broadcast?

```
if(!currentlyWithin(PrivacyZone))
{
transmitLocationdata();
}
```
I don't really understand what privacy issues there are? Entering in the privacy zone in the app?

garthvh, 2025-03-01:
> I might be oversimplifying it, but isn't it relatively easy to add a condition just before a location data packet is broadcast?

We are waiting for the results so adding privacy related features before we have those results is potentially problematic, needs to be done in a unified way.

garthvh, 2025-03-01:
Is a good feature, just dependent on some other in process work

Kealper, 2025-03-16:
Looking forward to progress on this, I often forget to switch my position precision back down to the usual "I'm in this rough area" setting I use while at home after I've been out in places where I want my exact position shared with others.

Something as simple as having, for example, 1-3 configurable "privacy zones" with a selectable precision where, if your exact position falls within that zone's precision radius, the final position transmitted gets masked with that zone's precision. It wouldn't have to save the actual exact position of the zone, just the center coordinates with the selected precision that will always give that specific precision radius, so even if the device is compromised, the saved zones won't ever divulge any more information than could already be obtained over the mesh while the user is in one of the zones.

b8b8, 2025-03-16:
> > Sports app Strava calls them "privacy zones". For reference, here are the options to choose from:
> 
> This could work with some additional features in place, the strava functionality depends on user accounts and encryption being in place for the location data.

Does that mean user accounts are coming to meshtastic?

flackr, 2025-08-05:
There's a few other settings that might be nice to change based on location boundaries. Of course there may be better ways to accomplish these things. For example,

1. When I'm at home, I want my device to be in CLIENT_MUTE mode whereas while away I would prefer it to be in CLIENT mode as repeating messages may be useful. Of course there may be better ways of doing that - e.g. perhaps a list of nodes that when I have a good signal to the device acts as CLIENT_MUTE under the assumption that this node is a better one to repeat messages.
2. When at home I'm also stationary so I'd rather enable "Compass north top". Though this would arguably be better done by having north be assumed when roughly stationary.