meshtastic/firmware #6183: [Feature Request]: 15 hops on Short Fast/Short Turbo in v3.x

mikecarper, 2025-02-28:
### Platform

Cross-Platform

### Description

Add another bit to the hop count only with the faster transmit times of the quickest 2 short presets. 10x+ quicker transmissions in comparison to LongFast so the extra hops should allow for better coverage with badly placed and configured nodes.

caveman99, 2025-03-01:
we don't have another bit we can use.

closed 2025-03-01 by caveman99

Talie5in, 2025-03-01:
@caveman what about extending into hex? 0-9A-F 
That should give "15"

mikecarper, 2025-03-01:
That's why it was a 3.x request

caveman99, 2025-03-01:
> [@caveman](https://github.com/caveman) what about extending into hex? 0-9A-F That should give "15" in the same space.

0-7 is 3 bits. 0-F is 4 bits, no amout of changing between decimal and hexadecimal will change that.

caveman99, 2025-03-01:
> That's why it was a 3.x request

the 6 and 7 hop counts are a waste already. 15 will not change a thing in that. I don't see this even on 3.0. We have looked at this extensively with Meshtaticator.

mikecarper, 2025-03-01:
@caveman99 in the Seattle area we repeatability get contacts at 6-7 hops out