meshtastic/firmware #6191: [Feature Request]: Add something to identify if a node can be messaged.

garthvh, 2025-03-01:
### Platform

Cross-Platform

### Description

Users are adding long strings of text or the 📵 emoji to long name to identify nodes that are unattended or can't be messaged.

Having this distinction available programmatically by using node info or the role used by a device would help client apps reduce the contact list and improve messaging.

The majority of MQTT connected nodes suffer from this problem as most private brokers of moderate size are downlink only so messaging is totally unavailable to those nodes but there is no way for a client app to know this.

Once we can better identify nodes that can successfully be messaged then the rolling node db functionality can prioritize retaining nodes (and their PKI keys) for nodes where messaging is actually available and should substantially reduce key mismatch errors.

GUVWAF, 2025-03-01:
Related to #5875.

fifieldt, 2025-03-02:
This is a good idea. We could perhaps automatically apply the "can't be mesaged" to router roles?

fifieldt, 2025-03-02:
App design note - nodes that can't be messaged can still be important targets for traceroute and position exchange.

Routers, and Tracker roles are still important on the map view.

garthvh, 2025-03-30:
Closed by https://github.com/meshtastic/protobufs/pull/667#pullrequestreview-2728042459

closed 2025-05-23 by garthvh