meshtastic/firmware #6195: [Feature Request]: CardKB Support with HUD UI

KC9SEB, 2025-03-01:
### Platform

ESP32

### Description

Currently CardKB does not seem to work with 2.6 / HUD UI. Older Alpha / Beta releases (Non HUD) work fine. The device to detected....but does not respond.

caveman99, 2025-03-01:
The InkHUD is a completely new UI much like the MUI for TFT devices. It lacks the freetext input feature. @todd-herbert will probably have a look in due time. For now if your require freetext entry please use the non-InkHUD version of the firmware.

Xaositek, 2025-03-03:
Adding in @tropho23 and @HarukiToreda

blueskies76, 2025-06-01:
I’ve noticed a growing trend of users adopting the Heltec T213, likely due to its larger screen. At the same time, I also built many  standalone CardKB-based devices for Meshtastic. I believe that adding support for a HUD-style UI with CardKB integration could significantly boost interest and adoption among users looking for more versatile standalone setups.

Xaositek, 2025-07-03:
With Meshtastic 2.7 - BaseUI should be able to accomplish this goal overall. I am not sure if Freetext has been built into InkHUD at this time.

todd-herbert, 2025-07-03:
<sub>(<https://github.com/meshtastic/firmware/blob/master/src/graphics/niche/InkHUD/docs/README.md#purpose>)</sub>
> InkHUD is a minimal UI for E-Ink devices. It displays the user's choice of info, as statically as possible, to minimize the amount of display refreshing.

My personal opinion is that these small E-Ink displays aren't particularly suitable for interactive use. I initially had no intention of adding keyboard support, however there's enough demand for it as a feature that I do feel obliged to add some form of free-text in future, even though it goes beyond the original scope of the UI. It's a low-priority job though; there are plenty of other more pressing things to sort first.

Both BaseUI and InkHUD are limited by their design choices, and neither is particularly suitable for E-Ink communicator-style devices. With some very deliberate design, I do think it would be possible to create from scratch a UI which truly does justice to that particular niche, but it would be a big outlay (several months of work?) for what is really a bit of an edge case.

closed 2025-08-26 by github-actions[bot]