meshtastic/firmware #6213: [Bug]: GPIO incorrect for Lilygo Lora32 2.1 1.6.1

edwar64896, 2025-03-03:
### Category

Other

### Hardware

T-Lora v2 1.6

### Firmware Version

2.5 (latest beta)

### Description

have connected a switch across IO12 and GND as per doc, however this doesn't appear to have any effect as a user button. Continuity and circiut both checked. Am currently checking code for this board to see where the GPIO is configured...

### Relevant log output

```Shell

```

edwar64896, 2025-03-03:
3ae85e2 appears to be the commit where BUTTON_PIN was nuked. I read somewhere that for the PIN_BUTTON configuration parameter to function, BUTTON_PIN must be defined...

edwar64896, 2025-03-03:
Further investigation: as this is my first foray into this repository, not entirely sure yet exactly where buttonGpio is getting handled for the Lilygo TLora32, but certainly, putting #define BUTTON_PIN 12 back into variant.h fixes the issue. I suspect though this isn't the end of the story.

edwar64896, 2025-03-03:
ok here we are...

https://github.com/meshtastic/firmware/blob/b6562e175fe90e0eeb069d6dda9e6a70ff18c096/src/ButtonThread.cpp#L31

Suspect that there are going to be more references like this meaning that really, BUTTON_PIN is going to have to be defined in variant.h for now unless someone wants to dig through the whole codebase to fix the dependency!

ndoo, 2025-03-04:
Seems like the logic that makes the user definition not work needs to be fixed instead. Because restoring this causes electrical noise to trigger the GPIO since it's floating (as expected) since the board has no pull up/down resistors on the pin (as it was never intended to be used this way).

todd-herbert, 2025-03-05:
The preferred way for tlora_v2_1_16 to add a user button would be by [setting config.device.button_gpio](https://meshtastic.org/docs/configuration/radio/device/#gpio-for-user-button).  It looks to me though like setting this value won't have any impact if the variant doesn't originally define a `BUTTON_PIN` or `USERPREFS_BUTTON_PIN`.

Looking back through git blame, this one might be my fault. Sorry about that..
> Fix button interrupt after light sleep (https://github.com/meshtastic/firmware/pull/3587)

> * Don't assume device has a button.. 

This doesn't seem to have impacted NRF52 devices, because they tend to define `PIN_BUTTON1` in variant.h, which is then assigned as `BUTTON PIN` here https://github.com/meshtastic/firmware/blob/f0f2cd0e0e3d2245bd02a20580232ca96bd3a86b/src/platform/nrf52/architecture.h#L104-L106

I think the reason we're noticing this now is: a few months ago, the decision was made that things like `BUTTON_PIN` shouldn't be defined in variant.h unless they're stock hardware on a board. That attitude definitely makes sense to me, but as you guys had already spotted, it seems like the ButtonThread logic needs some updating to accommodate this.

As @edwar64896 also noticed, the doc about [adding a button to LILYGO® TTGO Lora](https://meshtastic.org/docs/hardware/devices/lilygo/lora/gpio/#gpio-io12) also needs updating.

Are either of you keen to work on changes to ButtonThread, or would you like me to take a look at it?

edwar64896, 2025-03-09:
keen but about to jump into a 3 week project so will have bandwidth early April. If this isn't soon enough am happy to watch from the sidelines...

mmartin-bbw, 2025-03-21:
I am seeing the same thing. I guess this does not need further confirmation, but still, just wanted to say I am very much looking forward to seing this functionality restored.

ndoo, 2025-03-22:
Having looked at ButtonThread.cpp, this 1-line change should fix the issue, but I don't have a LoRa32 2.1 1.6.1 to test this weekend, the soonest I can test is the next working day.

Can anyone give the above change a try?

ndoo, 2025-03-22:
After compile-testing, I updated it to fix breaking the build, and updated main.cpp to ensure the GPIO gets initialized.  Pushing a new commit shortly.

ndoo, 2025-03-22:
Have a look at the commit above @todd-herbert but I think this just gets messy and no other variant seems to use USERPREFS_BUTTON_PIN anyway.

Perhaps the preference is to just get rid of USERPREFS_BUTTON_PIN and like in nRF52, define BUTTON_PIN but as (-1) for ESP32.

If that's the case, let me know and will rework the commit in that direction and open a PR.

ndoo, 2025-03-22:
I think defining GPIO as -1 by default on ESP32 works better, not sure if it will break variants without BUTTON_PIN defined at build time, will test on hardware in a few days.

https://github.com/meshtastic/firmware/pull/6371

closed 2025-03-23 by caveman99

caveman99, 2025-03-31:
Re-opening since the patch was reverted

IZ1IVA, 2025-04-30:
Hello, is anyone still working on this? Cheers

ndoo, 2025-06-29:
> Hello, is anyone still working on this? Cheers

https://github.com/meshtastic/firmware/pull/7170

I opened a PR to revert my earlier commit - would appreciate if you can checkout and build [my branch](https://github.com/ndoo/meshtastic-firmware/tree/revert-5535) to see if it solves the issue.

Please post any positive result on that thread and then I will mark the PR as ready.

ndoo, 2025-07-04:
> Hello, is anyone still working on this? Cheers

Hi there, did you manage to confirm if the PR will fix this?

IZ1IVA, 2025-07-04:
> Hi there, did you manage to confirm if the PR will fix this?

Hi! Thank you for taking the time to address this bug. Unfortunately I couldn't test your PR, as the only working LoRa32 is a remote node and I've yet to find the time to go there and upload the new firmware.
Could you possibily provide a build? That would also help me.

Cheers!

max2697, 2025-07-13:
Hello @ndoo!  I have a T3 v1.6.1 board and I’m experiencing an issue with the onboard button not working on two of my boards.

At first, I suspected wiring issues, but even after resoldering, the problem wasn’t resolved. I tested your firmware (target tlora-v2-1-1_6), but didn’t notice any change - the button on pin 12 still doesn’t respond immediately after flashing.

I also tried setting GPIO to pin 12 and 13 using the iPhone app, but no success there either.

Please let me know how I can assist with diagnosing this issue further.

ramajor, 2025-07-14:
@max2697 did you build from the revert-5535 branch? Because I just flashed this branch to my Lilygo and the button is working for me now on pin 12.

max2697, 2025-07-14:
@ramajor Yep… maybe something’s wrong with pin 12 on my board.

Today I even tried manually connecting ground to pin 12 - still no response. The reset pin works fine, though.

Is it possible to try using a different pin for the button? If so, which ones would be safe to test with?

ramajor, 2025-07-14:
@max2697 13,14,15,25 should all be good for a button.

ndoo, 2025-07-26:
> Also LilyGO LoRa32 v2.1 (1.6.1), Button needed to gpio12 and ground

so does the revert fix the issue? it should. getting some conflicting
reports

andrewsweet4, 2025-07-26:
I really don't know yet... Still my LilyGO LoRa32 v2.1 (1.6.1) button doesn't work. I think someone should merge the small change for the button in the main branch.

andrewsweet4, 2025-07-26:
Update: I just compiled the latest Version (2.7.4) with the addition of the two lines
#define BUTTON_PIN 12
and
#define BUTTON_NEED_PULLUP
in the variant.h file and the button works now, so yeah, that was a fix. I tested the code with 2 different LilyGO LoRa32 v2.1 (1.6.1) modules and works on both.

IZ1IVA, 2025-08-13:
> so does the revert fix the issue? it should. getting some conflicting
> reports

It does! Thank you very much for the fix, I hope it'll be merged soon.

BobDarlington, 2025-11-28:
> #define BUTTON_PIN 12
> #define BUTTON_NEED_PULLUP

Can confirm, just these two lines in variant.h worked for me with the latest code (as of today anyway).

tomalta, 2025-12-25:
I just installed the 2.7.16 alpha from the web flasher and no setting pin from the app or web app works.

gh0stless, 2026-10-30:
Still present in **v2.7.26**. Same workaround (`BUTTON_PIN 12` + `BUTTON_NEED_PULLUP` in `variant.h`) confirmed on two T3 V1.6.1 boards. Setting "GPIO for user button" in the app has no effect, because `InputBroker.cpp` only creates the user button thread under `#if defined(BUTTON_PIN)`.

The real problem is the user experience, not the code. Nobody reads every changelog. A button that worked yesterday is silently dead after a routine update, and the documented replacement (setting the GPIO in the app) does nothing. So people assume their hardware is broken. I assumed my boards were defective until I dug into the source, and judging by this thread I'm not the only one.

Until the app setting actually works without a compile-time `BUTTON_PIN`, please restore the define for this variant (e.g. merge #7170 or #6222). The internal pull-up keeps an unconnected GPIO12 stable, and it's harmless at boot because the internal pull-up isn't active during strapping. It can be removed again once `button_gpio` is handled properly.

PR #6371 (2025-03-23): [esp32] Define BUTTON_PIN (-1) by default, fixes #6213
ESP32 - Define BUTTON_PIN (-1) by default, fixes #6213 because button setup is always done in https://github.com/meshtastic/firmware/blob/1e4a0134e6ed6d455e54cd21f64232389280781b/src/main.cpp#L375-L392 and https://github.com/meshtastic/firmware/blob/1e4a0134e6ed6d455e54cd21f64232389280781b/src/ButtonThread.cpp#L31-L58 when the user preference for button pin is configured.
commit daa4186d (2025-03-23): [esp32] Define BUTTON_PIN (-1) by default, fixes #6213 (#6371)