meshtastic/firmware #6243: [Bug]: Blank display when press USER button on Heltec Wireless Tracker v1.2

violarulan, 2025-03-06:
### Category

Other

### Hardware

Heltec Wireless Tracker

### Firmware Version

2.5.20

### Description

Board: HTIT Tracker v1.2

I can flash the latest Beta firmware into the board normally. The board can boot and everything seems working.

After some time, when I press the USER button, the backlight goes up but there is no content. 
The BLE/LoRa can work normally meanwhile. The backlight goes off after the display timeout which is set in the App.

What I have tried:

- Fresh re-install: **_problem exists_**
- Remove the battery and try to use USB power: **_problem exists_**
- Test on another Tracker board (v1.2): **_problem exists_**
- Flash other firmware [https://github.com/lora-aprs/LoRa_APRS_Tracker](https://github.com/lora-aprs/LoRa_APRS_Tracker): **_the display works normally_**

I notice the supported version [https://flasher.meshtastic.org/](https://flasher.meshtastic.org/) is v1.1 - I'm using version v1.2. Is it relevant?

Test video: https://streamable.com/4t6i5c
Board image: https://imgur.com/a/g174vw8

Below is the log in which I pressed the USER button twice during the test. 

### Relevant log output

```Shell
INFO | 01:07:46 905 [GPS] GPS power state move from ACTIVE to HARDSLEEP
DEBUG | 01:07:46 905 [GPS] Publish pos@0:2, hasVal=0, Sats=0, GPSlock=0
DEBUG | 01:07:46 905 [GPS] onGPSChanged() pos@0 time=1741252066 lat=0 lon=0 alt=0
INFO | 01:07:46 905 [GPS] updatePosition LOCAL pos@0 time=1741252066 lat=0 lon=0 alt=0
DEBUG | 01:07:46 905 [GPS] Set local position: lat=0 lon=0 time=1741252066 timestamp=0
DEBUG | 01:07:46 905 [GPS] Node status update: 2 online, 3 total
DEBUG | 01:07:48 907 [Power] Battery: usbPower=0, isCharging=0, batMv=3747, batPct=53
INFO | 01:07:59 918 [Button] Loss of power in Powered
INFO | 01:07:59 918 [PowerFSM] Loss of power in Powered
DEBUG | 01:07:59 918 [PowerFSM] State: ON
INFO | 01:07:59 918 [Screen] Turn on screen
DEBUG | 01:08:08 927 [Power] Battery: usbPower=0, isCharging=0, batMv=3755, batPct=54
INFO | 01:08:14 934 [PowerFSM] Turn off screen
DEBUG | 01:08:28 947 [Power] Battery: usbPower=0, isCharging=0, batMv=3767, batPct=55
DEBUG | 01:08:48 967 [Power] Battery: usbPower=0, isCharging=0, batMv=3757, batPct=54
DEBUG | 01:09:08 987 [Power] Battery: usbPower=0, isCharging=0, batMv=3776, batPct=57
DEBUG | 01:09:28 1007 [Power] Battery: usbPower=0, isCharging=0, batMv=3754, batPct=54
INFO | 01:09:47 1026 [GPS] GPS power state move from HARDSLEEP to ACTIVE
DEBUG | 01:09:48 1027 [Power] Battery: usbPower=0, isCharging=0, batMv=3748, batPct=53
DEBUG | 01:10:08 1047 [Power] Battery: usbPower=0, isCharging=0, batMv=3735, batPct=51
INFO | 01:10:18 1057 [Button] Loss of power in Powered
INFO | 01:10:18 1057 [PowerFSM] Loss of power in Powered
DEBUG | 01:10:18 1057 [PowerFSM] State: ON
INFO | 01:10:18 1057 [Screen] Turn on screen
DEBUG | 01:10:22 1061 [Button] State: ON
DEBUG | 01:10:24 1063 [Button] State: ON
DEBUG | 01:10:28 1067 [Power] Battery: usbPower=0, isCharging=0, batMv=3726, batPct=50
INFO | 01:10:39 1078 [PowerFSM] Turn off screen
```

madeofstown, 2025-03-10:
Hmm, v1.2 is definitely a newer revision that what is supported by Meshtastic. I'm guessing Heltec did something like putting GPS power and backlight power on the same switch.

jlaughter, 2025-03-30:
Lemme add a "me too" to this.   Exact same behavior with two brand new HTIT v1.2 boards.

jlaughter, 2025-04-17:
Did further research on this.   Confirmed that if Display:Screen Timeout is set to 0, the device never enters the unrecoverable state where the backlight is on and the display is blank.  So it appears that the issue is with waking up after display timeout.

optikas, 2025-04-30:
Me too, after screen time out, it can wake up backlight, but show nothing.

Rockaflo, 2025-05-21:
Hello there!
I had the same problem with my v1.1 tracker., so i flashed the 2.6.8 Alpha firmware.
This solved my "empty display problem".
Hope this helps for yours too!

closed 2025-05-22 by thebentern

xBelladonna, 2025-06-23:
> I have a new Heltec Wireless Tracker (HTIT 1.2), immediately flashed it to 2.6.11 (first Meshtastic on this hardware) and I am seeing this exact problem: After some time, only the backlight comes back on when pressing the button, but nothing is displayed.

I had this issue before the update that fixed it, but even after I updated, it kept happening until I adjusted my display timeout. I'm not sure what it did, but it works as expected now even after changing the timeout back to the original value. Hopefully that fixes it for you too.

pdreker, 2025-06-23:
and sure enough: After modifying the display timeout (I set it to "infinite", if that matters), waiting a bit and setting it back to 10 minutes it now works as intended.

Thanks for the hint.

Jaymax15, 2025-07-21:
I have set my display to always on and this has resolved the issue but it drains battery quickly