meshtastic/firmware #6244: [Bug]: On Heltec T114 is a random Bluetooth-Connect-Code shown on the screen. This also if Bluetooth is configured als fixed value

Stephan-2025, 2025-03-06:
### Category

BLE

### Hardware

Heltec Mesh Node T114

### Firmware Version

2.5.20

### Description

On Heltec T114 is a random Bluetooth-Connect-Code shown on the screen. This also if Bluetooth is configured als fixed value. The problem appears only, if a screen is attached at the T114. If not i works as designed

### Relevant log output

```Shell
I have no Log, sorry
```

Stephan-2025, 2025-03-06:
I´ve set not only the fixed pin, i also set the Pairing mode on "FIXED_PIN" - but the Node still shows and use a random pin on the display

Stephan-2025, 2025-03-12:
I´ve tried to reproduce, now there work both Bluetooth-Keys, that one who was set as fixed as well as that one (it´s different) shown on the screen. So for that it´s no more a bug ... more a special feature :) ;)

fifieldt, 2025-03-21:
Glad you managed to fix it, thanks for taking the time to send in the report :)

closed 2025-03-21 by fifieldt