meshtastic/firmware #6262: [Feature Request]: Configurable hops for Telemetry

ShakataGaNai, 2025-03-07:
### Platform

Cross-Platform

### Description

In at least the Telemetry module, and possible others, it would be nice to have the ability to configure the number of hops. The use case would be support more sensor based nodes inside of a larger mesh, without otherwise impacting larger mesh functionality.

The idea is that you could deploy node that is sensor focused with a zero hop limit so as to not be broadcast across hundreds of miles of mesh. It's data either makes it to the local MQTT uplink (if that's your goal), or your "main" node in the area, or it dies. It allows someone using MQTT to collect more data, while not being an undue burden on the larger mesh. 

Yes, for a dedicated sensor-only node you might simply set the nodes hops to Zero globally, but there are many cases (ex: vehicle node, or handheld node) where you still want to use it for normal communications, but don't really care if the telemetry data gets farther along. Especially with the addition of sensors for health monitoring.

thebentern, 2025-03-08:
I like the concept here as a traffic reduction technique for lower-priority data. I'm wondering if we can get away with a `lora.hop_limit - 1` approach by-default for telemetry unless you are in a role where that data becomes more important such as `ROUTER` or `SENSOR`, instead of providing explicit hop limit per portnum, which feels a like it could turn into a bit of a pandoras box of configuration knobs spilling out to all of the ports eventually.

GUVWAF, 2025-03-08:
> I'm wondering if we can get away with a lora.hop_limit - 1 approach by-default for telemetry

I would not be in favor of this, as it makes things unpredictable. Maybe someone configures their hop limit already to accommodate for a certain reach, but then it doesn't work for telemetry. 

In general configuring the broadcast interval will have a larger impact on channel utilization and I think the dynamic scaling (#5264) introduced not so long ago will help a lot also. Maybe eventually we should also make DeviceTelemetry opt-in, although it's now the only automatic broadcast outside of NodeInfo (and Position if you have a valid position), which is also handy for node discovery.

Indeed creating hop limit settings per portnum sounds cumbersome. I think what could maybe considered eventually -when 2.6 is more widely deployed- is to configure a certain NodeNum to send the telemetry to, instead of broadcasting it.

thebentern, 2025-03-10:
> I would not be in favor of this, as it makes things unpredictable. Maybe someone configures their hop limit already to accommodate for a certain reach, but then it doesn't work for telemetry.

Makes sense. Once we get more networks upgraded to 2.6, we'll have a better baseline to compare against, as right now we still have a lot nodes on older versions for various reasons.

closed 2025-05-02 by github-actions[bot]