meshtastic/firmware #6265: [Feature Request]: Allow S&F on default channel for ver 2.6+

RCGV1, 2025-03-08:
### Platform

Cross-Platform

### Description

The S&F module uses directed messages to send down messages a user missed. With the new next hop routing, the potential channel util increase is reduced because necessary retransmissions are removed.

garthvh, 2025-03-14:
Store and forward is just too much traffic for long fast, next hop routing or not.  This is a feature for faster presets and should be a good way to incentivize moving to faster presets

Woutvstk, 2025-03-16:
The S&F could also be limited to one or zero hops, but that somewhat limits its functionality. You'd have to be close to the S&F node then. 

When a strong direct connection is detected, you could switch server and client to a faster preset (and hop limit to 0 hops, so no retransmissions) to send a large message history, then switch back.

RCGV1, 2025-03-16:
I think we can allow SF to populate the ch util a bit more, messages are extremely important to users. I also plan on implementing ways to improve the efficiency of SF like #6332

garthvh, 2025-03-17:
Problem is it is not a little bit, it winds up being a flood of messages that almost immediately exceeds the channel utilization limits.

closed 2025-05-09 by github-actions[bot]