meshtastic/firmware #6269: [Bug]: GPS position update stops when hiking

feh123, 2025-03-08:
### Category

Other

### Hardware

Seeed Card Tracker T1000-E

### Firmware Version

2.5.22

### Description

This has been a problem I have seen for many months. I have no data to attach but I think the rak4631 devices I have also give me the same problem. I should say at the start that this could be down to me! The issue: I go for a hike hoping to see my position update with smart position. In fact after a period of time (usually after at least one hour) GPS updates stop. 

These attached files show the issue. The position log shows two circular walks I did and in both updates stop before I complete the walk. They are artificially joined if you convert the csv file to

[FEH Position Log.csv](https://github.com/user-attachments/files/19144856/FEH.Position.Log.csv)

[FEH Device Metrics Log.csv](https://github.com/user-attachments/files/19144857/FEH.Device.Metrics.Log.csv)

gpx - please ignore. I did arrive back at the start point but according to the data I stopped about 60-70% of the way round. I also attach the device metrics log. As you can see I am in the UK so I have the 10% cycle time. I do wonder if this is the source of my problem. Maybe someone can check if I am hitting the hourly limit? I am not sure how to add up all the airtime I use. My device settings are: MediumFast, precise location on two channels, Local Only, smart position enabled with Broadcast interval 2hr, Minimum interval 3 min, Minimum distance 50 m, GPS Update interval 2 mins (for the first hike, 5 mins for the second hike), flags only altitude. I am not using any power save settings. I use the latest iOS App from the App store. I only had one device working, it's on the outside of my pack just hanging inside a pocket. The routes were very flat and I always had a full view of the sky. The battery level was always good. Any thoughts very much appreciated as, at present, both the T1000e and the rak4631 devices I have are quite useless for hiking. I am hoping there is something I am missing with my settings! Thanks! 

### Relevant log output

```Shell

```

fifieldt, 2025-03-09:
Hi @feh123 , thanks for writing in, and thanks for using Meshtastic!

Just to confirm the definition of "GPS updates stop", does this mean:
* when you check later in the CVS, no extra lines are recorded
* when you look at the map in the IoS app while connected by Bluetooth during your hike, your GPS location is not close to where you are
* your Rak4631 sees a bad location for your T1000E device and vice versa, when looking at the coordinates displayed on the list of nodes.
* ?

Are there any differences you see between your T1000E and Rak4631?

feh123, 2025-03-09:
Hi @fifieldt thanks for your response. To answer your questions:

1 GPS updates stop: yes there are no more lines in the csv - nothing is being recorded.

2 I did not look at the map on my phone during the hikes! Sorry! It may seem strange but I was just recording the hikes as a test of device. To try to prove to myself it was useful to carry a T1000e. The points that are there are very accurate and follow exactly the route I took. 

3 Rak v T1000e. I mentioned the rak4631 only to show that I thought this issue was present on a different type of device. My rak4631 devices I have had to be "made" by me with a large battery (to cover the gps and display demands), a big gps antenna (to help get a better signal for the rather poor rak gps) and a box to hold it all. They are rather heavy, cannot be kept in an outer pocket of my pack (hence don't get a good sky view) and when the T1000e came out I stopped using them.  So I don't have "hiking" comparisons with the T1000e. At home side by side the rak position data is not as good as that from the T1000e - the rak's positions show a bigger variation for a stationary device. Another reason I stopped using them.

Re the ? I had two ideas. I wanted to see what the smart position values did, hence the change in update interval. And I wanted to test the reliability of the tracking. It maybe surprising but I have never recorded a good meshtastic gps track! The rak's have never worked that well - I think due to the poor gps being inside a pack.
I had two theories about why the gps update stops - the device is going to sleep in some way (I don't know what would activate this) or the EU 10% airtime cycle was kicking in. Given the fact that the gps packets were seen initially and stopped after 1-2 hours I was thinking my experience fitted the airtime cycle. Hence I added the Device Metrics log to see if I was hitting the limit. 

Or it could be due to something else! Can you tell from the device metrics if I was hitting the cycle limit? I don't know how the device metrics log is calculating these %'s so I don't like to try to do this myself. I have done everything I can think of to reduce the number of packets I send - all that is left is to further attenuate the smart position values to reduce the gps packets. Say 100 m every 10 minutes? As anyone investigated the "best" smart position values for packet rate versus position with EU_868? Thanks again for you reply.

feh123, 2025-03-14:
The T1000e's are working fine now thanks to @b8b8 and @cynfab - lots of appreciation for their help and kudos to them. As expected it was my fault - I had got into the habit of closing the App and disconnecting from the device. Somehow I had thought that the device did not need a bluetoothed phone. You can check out the details on Discord at #gnss.

closed 2025-03-14 by feh123