meshtastic/firmware #6274: [Bug]: T-Deck Lilygo

xXxCryxXx, 2025-03-09:
### Category

Other

### Hardware

T-Deck

### Firmware Version

2.6.0.f7afa9a

### Description

T-Deck is never waking up after 10 min of "sleep" with screen turned off. After turning it on via the Key´s or via Drackball it keeps getting freezd or Crashed. It turns many Times on an Off after turning it off via switch it is all okay. Also the Keyboard is turning on and off also if i receive a message it blinks up or even if a do a Trace route to the T-Deck the Keyboard flashes up. At this moment i cant you the T-Deck only 5 minutes after that amount of time it turns off or starts lagging dosent matter witch view i am on the GUI

### Relevant log output

```Shell

```

caveman99, 2025-03-23:
Is this still happening witt 2.6.2?

mverch67, 2025-03-28:
Do you have by chance attached the T-Deck to a Mac computer via USB while this happens?