meshtastic/firmware #6278: [Bug]: All mesh tab envs build fail

PJ5168, 2025-03-10:
### Category

Other

### Hardware

Other

### Firmware Version

2.6.0

### Description

All envs of mesh-tab build fail.
The build process will try to build bluetooth which is excluded in the ini.

### Relevant log output

```Shell

```

mverch67, 2025-03-11:
Seems you're doing something wrong. It's building properly.

```
Successfully created esp32s3 image.
esp32_create_combined_bin([".pio/build/mesh-tab-3-5-IPS-resistive/firmware.bin"], [".pio/build/mesh-tab-3-5-IPS-resistive/firmware.elf"])
Generating combined binary for serial flashing
    Offset | File
 -  0x0000 | /home/manuel/Documents/PlatformIO/Projects/meshtastic-latest/.pio/build/mesh-tab-3-5-IPS-resistive/bootloader.bin
 -  0x8000 | /home/manuel/Documents/PlatformIO/Projects/meshtastic-latest/.pio/build/mesh-tab-3-5-IPS-resistive/partitions.bin
 -  0xe000 | /home/manuel/.platformio/packages/framework-arduinoespressif32/tools/partitions/boot_app0.bin
 - 0x10000 | /home/manuel/Documents/PlatformIO/Projects/meshtastic-latest/.pio/build/mesh-tab-3-5-IPS-resistive/firmware.bin
Using esptool.py arguments: --chip esp32s3 merge_bin -o /home/manuel/Documents/PlatformIO/Projects/meshtastic-latest/.pio/build/mesh-tab-3-5-IPS-resistive/firmware.factory.bin --flash_mode dio --flash_freq 80m --flash_size 16MB 0x0000 /home/manuel/Documents/PlatformIO/Projects/meshtastic-latest/.pio/build/mesh-tab-3-5-IPS-resistive/bootloader.bin 0x8000 /home/manuel/Documents/PlatformIO/Projects/meshtastic-latest/.pio/build/mesh-tab-3-5-IPS-resistive/partitions.bin 0xe000 /home/manuel/.platformio/packages/framework-arduinoespressif32/tools/partitions/boot_app0.bin 0x10000 /home/manuel/Documents/PlatformIO/Projects/meshtastic-latest/.pio/build/mesh-tab-3-5-IPS-resistive/firmware.bin
esptool.py v4.8.1
SHA digest in image updated
Wrote 0x2e1380 bytes to file /home/manuel/Documents/PlatformIO/Projects/meshtastic-latest/.pio/build/mesh-tab-3-5-IPS-resistive/firmware.factory.bin, ready to flash to offset 0x0
================================================================================================== [SUCCESS] Took 171.37 seconds ==================================================================================================

Environment                 Status    Duration
--------------------------  --------  ------------
mesh-tab-3-5-IPS-resistive  SUCCESS   00:02:51.373
=================================================================================================== 1 succeeded in 00:02:51.373 ===================================================================================================
 *  Terminal will be reused by tasks, press any key to close it. 
```

closed 2025-03-11 by mverch67

PJ5168, 2025-03-12:
2.6.1.7 tft related builds still fail. Others are OK.
Tested:
1. T-deck-tft : 
.pio/libdeps/t-deck-tft/LovyanGFX/src/lgfx/v1/LGFXBase.hpp:1446:17: note: code may be misoptimized unless -fno-strict-aliasing is used
lto-wrapper.exe: fatal error: CreateProcess: No such file or directory
compilation terminated.
c:/users/z840/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: error: lto-wrapper failed
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\t-deck-tft\firmware.elf] Error 1
====================================================================================== [FAILED] Took 230.28 seconds ======================================================================================

Environment    Status    Duration
-------------  --------  ------------
t-deck-tft     FAILED    00:03:50.278
================================================================================= 1 failed, 0 succeeded in 00:03:50.278 =================================================================================
2.  Mesh-tab-3.5 : 
*** [.pio\build\mesh-tab-3-5-IPS-capacitive\src\BluetoothCommon.cpp.o] Error 1
xtensa-esp32s3-elf-g++: error: CreateProcess: No such file or directory
*** [.pio\build\mesh-tab-3-5-IPS-capacitive\src\ButtonThread.cpp.o] Error 1
====================================================================================== [FAILED] Took 97.83 seconds ======================================================================================

Environment                  Status    Duration
---------------------------  --------  ------------
mesh-tab-3-5-IPS-capacitive  FAILED    00:01:37.834
================================================================================= 1 failed, 0 succeeded in 00:01:37.834 =================================================================================

mverch67, 2025-03-12:
Did you try building using WSL?