meshtastic/firmware #6282: [Bug]: RAK4631 doesn't do TLS over Ethernet

kstrauser, 2025-03-11:
### Category

Other

### Hardware

Rak4631

### Firmware Version

2.5.20.4c97351

### Description

I have a RAK4631 radio with RAK13800 Ethernet. I also have an MQTT server that supports both plaintext (port 1883) and TLS (port 8883) connections.

Regardless of the value of `mqtt.tls_enabled`, the radio always connects to the server's plaintext port 1883. I would've expected to connect to TLS port 8883.

### Relevant log output

```Shell

```

garthvh, 2025-03-11:
There is no web server on a NRF board, I am pretty sure ssl is not supported.

closed 2025-03-11 by thebentern

kstrauser, 2025-03-11:
Is TLS only compiled in if the board has a web server?

kstrauser, 2025-03-17:
That's exactly it, @randshell. I've configured the radio to connect to the MQTT server with TLS, but it's silently downgraded to plaintext.

Also, the server is "local" in the "not the default shared server" sense. It has TLS configured with a legit signed certificate. For testing just now, I changed its MQTT config to:

```
  mqtt:
    address: mqtt.meshtastic.org
    enabled: true
    encryptionEnabled: true
    mapReportSettings:
      positionPrecision: 32
    mapReportingEnabled: true
    password: large4cats
    root: msh/US
    username: meshdev
    tlsEnabled: true
```

After the reboot, here's what that connection looks like in my logs:

```
INFO  | ??:??:?? 10 [mqtt] Connect directly to MQTT server mqtt.meshtastic.org, port: 1883, username: meshdev, password: large4cats
```

So even with the default public server, `tlsEnabled` is ignored. I agree that I'd rather see an error message than it silently downgrading to plaintext.

garthvh, 2025-03-17:
Tls is not supported by the public server, or via the client proxy, neither is an option on iOS and the other clients could do the same. Mqtt on a private broker can be configured but is pretty unreliable.

ianmcorvidae, 2025-03-17:
Most of these MQTT settings are now validated and should throw errors instead of connecting insecurily, starting with 2.5.23 I believe. As of the following commit, anyway: https://github.com/meshtastic/firmware/commit/3b0232de1b

It's also easy to tell if your node is running firmware including that commit because it changed the log message from saying "Connect directly to MQTT server" to say "Connecting directly to MQTT server".