meshtastic/firmware #6293: [Bug]: 2.6.1.7 tft related build fail

PJ5168, 2025-03-12:
### Category

Other

### Hardware

T-Deck

### Firmware Version

2.6.1.7

### Description

Fresh git and 2.6.1.7 with fresh installed latest VSC, PIO :
Following the build instructions on the web.
Tft related build fail. Non-tft build seems OK.

### Relevant log output

```Shell

```

PJ5168, 2025-03-12:
Tested :

T-deck-tft :
.pio/libdeps/t-deck-tft/LovyanGFX/src/lgfx/v1/LGFXBase.hpp:1446:17: note: code may be misoptimized unless -fno-strict-aliasing is used
lto-wrapper.exe: fatal error: CreateProcess: No such file or directory
compilation terminated.
c:/users/z840/.platformio/packages/toolchain-xtensa-esp32s3/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: error: lto-wrapper failed
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\t-deck-tft\firmware.elf] Error 1
====================================================================================== [FAILED] Took 230.28 seconds ======================================================================================
Environment Status Duration

t-deck-tft FAILED 00:03:50.278
================================================================================= 1 failed, 0 succeeded in 00:03:50.278 =================================================================================

Mesh-tab-3.5 :
*** [.pio\build\mesh-tab-3-5-IPS-capacitive\src\BluetoothCommon.cpp.o] Error 1
xtensa-esp32s3-elf-g++: error: CreateProcess: No such file or directory
*** [.pio\build\mesh-tab-3-5-IPS-capacitive\src\ButtonThread.cpp.o] Error 1
====================================================================================== [FAILED] Took 97.83 seconds ======================================================================================
Environment Status Duration

mesh-tab-3-5-IPS-capacitive FAILED 00:01:37.834
================================================================================= 1 failed, 0 succeeded in 00:01:37.834 =================================================================================

PJ5168, 2025-03-12:
3. seeed-sensecap-indicator-tft fail
4. native-tft fail

PJ5168, 2025-03-12:
Non-tft fail as well : 
-DAPP_VERSION=2.6.1
-DAPP_VERSION_SHORT=2.6.1
-DAPP_ENV=seeed-sensecap-indicator
-DUSERPREFS_TZ_STRING=\"tzplaceholder                                         \"
Compiling .pio\build\seeed-sensecap-indicator\lib451\LovyanGFX\lgfx\Fonts\IPA\lgfx_font_japan.c.o
Compiling .pio\build\seeed-sensecap-indicator\lib451\LovyanGFX\lgfx\Fonts\efont\lgfx_efont_cn.c.o
xtensa-esp32s3-elf-gcc: error: CreateProcess: No such file or directory
Compiling .pio\build\seeed-sensecap-indicator\lib451\LovyanGFX\lgfx\Fonts\efont\lgfx_efont_ja.c.o
xtensa-esp32s3-elf-gcc: error: CreateProcess: No such file or directory
*** [.pio\build\seeed-sensecap-indicator\lib451\LovyanGFX\lgfx\Fonts\IPA\lgfx_font_japan.c.o] Error 1
*** [.pio\build\seeed-sensecap-indicator\lib451\LovyanGFX\lgfx\Fonts\efont\lgfx_efont_cn.c.o] Error 1
xtensa-esp32s3-elf-gcc: error: CreateProcess: No such file or directory
*** [.pio\build\seeed-sensecap-indicator\lib451\LovyanGFX\lgfx\Fonts\efont\lgfx_efont_ja.c.o] Error 1

PJ5168, 2025-03-12:
Ok to build : 
1. M5stack core
2. Heltec T190

caveman99, 2025-03-12:
You can not build TFT firmware on windows. It's a size limitation of the operating system command line. There is no known fix or workaround. Turning off the lto means your binaries get considerably bigger.

Try building in WSL on Windows.

closed 2025-03-12 by caveman99