meshtastic/firmware #6294: [Bug]: T-Echo 433MHz Error #3 after 2.5.20

skyboysea, 2025-03-12:
### Category

Hardware Compatibility

### Hardware

T-Echo

### Firmware Version

2.5.21.447533a

### Description

On a T-Echo 433, any release after 2.5.20 after flashing gives "Fatal Error #3". A factory reset before installing the FW does not solve the issue.

### Relevant log output

```Shell

```

fifieldt, 2025-03-16:
Fatal error 3 means no radio was detected.

A log file would be very helpful to work out why. Connect to USB, go to Open Serial Monitor on HTTPS://flasher.meshtastic.org

skyboysea, 2025-03-17:
This is the log from FW 2.5.21

`INFO  | 04:09:05 9 Start meshradio init
INFO  | 04:09:05 9 Radio freq=433.875, config.lora.frequency_offset=0.000
INFO  | 04:09:05 9 Set radio: region=EU_433, name=LongFast, config=0, ch=3, power=22
INFO  | 04:09:05 9 myRegion->freqStart -> myRegion->freqEnd: 433.000000 -> 434.000000 (1.000000 MHz)
INFO  | 04:09:05 9 numChannels: 4 x 250.000kHz
INFO  | 04:09:05 9 channel_num: 4
INFO  | 04:09:05 9 frequency: 433.875000
INFO  | 04:09:05 9 Slot time: 77 msec
INFO  | 04:09:05 9 Set radio: final power level=22
INFO  | 04:09:06 9 SX126x init result -2
WARN  | 04:09:06 9 No SX1262 radio
DEBUG | 04:09:06 9 SX126xInterface(cs=24, irq=20, rst=25, busy=17)
DEBUG | 04:09:06 9 SX126X_DIO3_TCXO_VOLTAGE not defined, not using DIO3 as TCXO reference voltage
INFO  | 04:09:06 9 Start meshradio init
INFO  | 04:09:06 9 Radio freq=433.875, config.lora.frequency_offset=0.000
INFO  | 04:09:06 9 Set radio: region=EU_433, name=LongFast, config=0, ch=3, power=22
INFO  | 04:09:06 9 myRegion->freqStart -> myRegion->freqEnd: 433.000000 -> 434.000000 (1.000000 MHz)
INFO  | 04:09:06 9 numChannels: 4 x 250.000kHz
INFO  | 04:09:06 9 channel_num: 4
INFO  | 04:09:06 9 frequency: 433.875000
INFO  | 04:09:06 9 Slot time: 77 msec
INFO  | 04:09:06 9 Set radio: final power level=22
INFO  | 04:09:06 9 SX126x init result -707
WARN  | 04:09:06 9 No SX1268 radio
ERROR | 04:09:06 9 NOTE! Record critical error 3 at src/main.cpp:1112
INFO  | 04:09:06 9 PowerFSM init, USB power=1
DEBUG | 04:09:06 9 State: BOOT`

To compare this is from FW 2.5.20

`INFO  | 04:17:57 6 Start meshradio init
INFO  | 04:17:57 6 Radio freq=433.875, config.lora.frequency_offset=0.000
INFO  | 04:17:57 6 Set radio: region=EU_433, name=LongFast, config=0, ch=3, power=22
INFO  | 04:17:57 6 myRegion->freqStart -> myRegion->freqEnd: 433.000000 -> 434.000000 (1.000000 MHz)
INFO  | 04:17:57 6 numChannels: 4 x 250.000kHz
INFO  | 04:17:57 6 channel_num: 4
INFO  | 04:17:57 6 frequency: 433.875000
INFO  | 04:17:57 6 Slot time: 77 msec
INFO  | 04:17:57 6 Set radio: final power level=22
INFO  | 04:17:58 6 SX126x init result -2
WARN  | 04:17:58 6 No SX1262 radio
DEBUG | 04:17:58 6 SX126xInterface(cs=24, irq=20, rst=25, busy=17)
DEBUG | 04:17:58 6 SX126X_DIO3_TCXO_VOLTAGE defined, using DIO3 as TCXO reference voltage at 1.800000 V
INFO  | 04:17:58 6 Start meshradio init
INFO  | 04:17:58 6 Radio freq=433.875, config.lora.frequency_offset=0.000
INFO  | 04:17:58 6 Set radio: region=EU_433, name=LongFast, config=0, ch=3, power=22
INFO  | 04:17:58 6 myRegion->freqStart -> myRegion->freqEnd: 433.000000 -> 434.000000 (1.000000 MHz)
INFO  | 04:17:58 6 numChannels: 4 x 250.000kHz
INFO  | 04:17:58 6 channel_num: 4
INFO  | 04:17:58 6 frequency: 433.875000
INFO  | 04:17:58 6 Slot time: 77 msec
INFO  | 04:17:58 6 Set radio: final power level=22
INFO  | 04:17:58 6 SX126x init result 0
INFO  | 04:17:58 6 Frequency set to 433.875000
INFO  | 04:17:58 6 Bandwidth set to 250.000000
INFO  | 04:17:58 6 Power output set to 22
DEBUG | 04:17:58 6 Current limit set to 140.000000
DEBUG | 04:17:58 6 Current limit set result 0
DEBUG | 04:17:58 6 Set DIO2 as RF switch, result: 0
DEBUG | 04:17:58 6 SX126X_RXEN not defined, defaulting to RADIOLIB_NC
DEBUG | 04:17:58 6 SX126X_TXEN not defined, defaulting to RADIOLIB_NC
DEBUG | 04:17:58 6 Use MCU pin -1 as RXEN and pin -1 as TXEN to control RF switching
INFO  | 04:17:58 6 Set RX gain to boosted mode; result: 0
INFO  | 04:17:58 6 SX1268 init success
DEBUG | 04:17:58 6 LoRA bitrate = 118.394310 bytes / sec
INFO  | 04:17:58 6 PowerFSM init, USB power=1
DEBUG | 04:17:58 6 State: BOOT`

fifieldt, 2025-03-17:
Thanks, very useful. That error is

`RADIOLIB_ERR_SPI_CMD_FAILED   (-707)
 	[SX126x](https://jgromes.github.io/RadioLib/class_s_x126x.html) failed to execute SPI command. Often this means that the module is trying to use TCXO while XTAL is connected (or vice versa). Make sure your crystal setup (e.g. TCXO reference voltage) matches your hardware by setting "tcxoVoltage" to 0 when using XTAL module, or to appropriate value when using TCXO module.`

fifieldt, 2025-03-17:
@NomDeTom - you're an XTAL vs TCXO expert - any ideas on what happening with this t-echo?

NomDeTom, 2025-03-20:
@fifieldt 
> [@NomDeTom](https://github.com/NomDeTom) - you're an XTAL vs TCXO expert - any ideas on what happening with this t-echo?

Something seemed to change after 1.5.20 - I've seen a few reports of failures. I wonder what it would do with a later version than 2.5.23 - in [pr6110](https://github.com/meshtastic/firmware/pull/6110) @thebentern added some extra bits back in to change some things with regard to `TCXO_Optional`

The other thing I wonder is if its tripping over this strange #define https://github.com/meshtastic/firmware/blob/f8ad02aab3cf92d4d3fa1b65ed8d41c1807f5766/variants/t-echo/variant.h#L134C1-L136C8

Anyway, hopefully something fixes them!

caveman99, 2025-03-23:
please test if 2.6.2 works again.

skyboysea, 2025-03-24:
The problem still exist on 2.6.2.

skyboysea, 2025-03-30:
No resolution in 2.6.4 either.

skyboysea, 2025-05-04:
Still present in 2.6.6

Logicbloke, 2025-05-10:
Experiencing the same issue on 2.6.7

I'm using LilyGo T-Echo nRF52840 and Meshtastic 2.6.7, the issue happens regardless if using inkHUD or not.

```
INFO | 21:08:26 8 Turn on screen
DEBUG | 21:08:26 8 haveGlyphs=1
DEBUG | 21:08:26 8 determineMode():
DEBUG | 21:08:26 8 refresh=FAST, reason=FLAGGED_DEMAND_FAST, frameFlags=0xb
DEBUG | 21:08:26 8 Update E-Paper
DEBUG | 21:08:27 9 done
DEBUG | 21:08:27 9 waypoint wants a UI Frame
DEBUG | 21:08:27 9 canned wants a UI Frame
DEBUG | 21:08:27 9 SX126xInterface(cs=24, irq=20, rst=25, busy=17)
DEBUG | 21:08:27 9 SX126X_DIO3_TCXO_VOLTAGE defined, using DIO3 as TCXO reference voltage at 1.800000 V
INFO | 21:08:27 9 Start meshradio init
INFO | 21:08:27 9 Radio freq=433.875, config.lora.frequency_offset=0.000
INFO | 21:08:27 9 Set radio: region=EU_433, name=LongFast, config=0, ch=3, power=12
INFO | 21:08:27 9 myRegion->freqStart -> myRegion->freqEnd: 433.000000 -> 434.000000 (1.000000 MHz)
INFO | 21:08:27 9 numChannels: 4 x 250.000kHz
INFO | 21:08:27 9 channel_num: 4
INFO | 21:08:27 9 frequency: 433.875000
INFO | 21:08:27 9 Slot time: 28 msec
INFO | 21:08:27 9 Set radio: final power level=12
INFO | 21:08:28 9 SX126x init result -2
WARN | 21:08:28 9 No SX1262 radio
DEBUG | 21:08:28 9 SX126xInterface(cs=24, irq=20, rst=25, busy=17)
DEBUG | 21:08:28 9 SX126X_DIO3_TCXO_VOLTAGE not defined, not using DIO3 as TCXO reference voltage
INFO | 21:08:28 9 Start meshradio init
INFO | 21:08:28 9 Radio freq=433.875, config.lora.frequency_offset=0.000
INFO | 21:08:28 9 Set radio: region=EU_433, name=LongFast, config=0, ch=3, power=12
INFO | 21:08:28 9 myRegion->freqStart -> myRegion->freqEnd: 433.000000 -> 434.000000 (1.000000 MHz)
INFO | 21:08:28 9 numChannels: 4 x 250.000kHz
INFO | 21:08:28 9 channel_num: 4
INFO | 21:08:28 9 frequency: 433.875000
INFO | 21:08:28 9 Slot time: 28 msec
INFO | 21:08:28 9 Set radio: final power level=12
INFO | 21:08:28 9 SX126x init result -707
WARN | 21:08:28 9 No SX1268 radio
ERROR | 21:08:28 9 NOTE! Record critical error 3 at src/main.cpp:1281
INFO | 21:08:28 9 PowerFSM init, USB power=1
DEBUG | 21:08:28 9 State: BOOT
DEBUG | 21:08:28 9 [Power] Battery: usbPower=1, isCharging=1, batMv=4706, batPct=100
DEBUG | 21:08:28 9 [Screen] Screen: Started...
DEBUG | 21:08:28 9 [Screen] Screen: Critical error 3!
DEBUG | 21:08:28 9 [Screen] haveGlyphs=1
DEBUG | 21:08:28 9 [Screen] determineMode():
DEBUG | 21:08:28 9 [Screen] refresh=SKIPPED, reason=FRAME_MATCHED_PREVIOUS, frameFlags=0x3
DEBUG | 21:08:28 9 [GPS] Probe for GPS at 9600
DEBUG | 21:08:28 9 [GPS] Trying $PDTINFO (Unicore Family)...
DEBUG | 21:08:28 10 [GPS] Trying $PCAS06,1*1A (ATGM33xx Family)...
DEBUG | 21:08:29 10 [GPS] Trying $PAIR021*39 (Airoha Family)...
DEBUG | 21:08:30 11 [GPS] Trying $PQTMVERNO*58 (LC86)...
DEBUG | 21:08:30 12 [GPS] Trying $PCAS06,0*1B (L76K)...
INFO | 21:08:30 12 [GPS] L76K detected
DEBUG | 21:08:31 13 [GPS] Publish pos@0:2, hasVal=0, Sats=0, GPSlock=0
DEBUG | 21:08:31 13 [GPS] No GPS lock
DEBUG | 21:08:31 13 [GPS] onGPSChanged() pos@0 time=0 lat=0 lon=0 alt=0
INFO | 21:08:31 13 [GPS] updatePosition LOCAL pos@0 time=0 lat=0 lon=0 alt=0
DEBUG | 21:08:31 13 [GPS] Set local position: lat=0 lon=0 time=0 timestamp=0
DEBUG | 21:08:31 13 [GPS] Node status update: 0 online, 1 total
DEBUG | 21:08:31 13 [GPS] NMEA GPS time 2025-05-10 21:10:22 age 0
DEBUG | 21:08:31 13 [GPS] Upgrade time to quality GPS
DEBUG | 21:10:22 13 [GPS] PCF8563_RTC setDateTime 2025-05-10 21:10:22 (1746911422)
DEBUG | 21:10:22 13 [GPS] Read RTC time from PCF8563 getDateTime as 2025-05-10 21:10:22 (1746911422)
DEBUG | 21:10:22 13 [GPS] Publish pos@0:2, hasVal=0, Sats=0, GPSlock=0
DEBUG | 21:10:22 13 [GPS] onGPSChanged() pos@0 time=1746911422 lat=0 lon=0 alt=0
INFO | 21:10:22 13 [GPS] updatePosition LOCAL pos@0 time=1746911422 lat=0 lon=0 alt=0
DEBUG | 21:10:22 13 [GPS] Set local position: lat=0 lon=0 time=1746911422 timestamp=0
DEBUG | 21:10:22 13 [GPS] Node status update: 1 online, 1 total
INFO | 21:10:22 13 [RangeTest] Range Test Module - Disabled
DEBUG | 21:10:22 13 [PowerFSM] Init NRF52 Bluetooth
INFO | 21:10:22 13 [PowerFSM] Init the Bluefruit nRF52 module
INFO | 21:10:22 13 [PowerFSM] Bluetooth pin set to '550895'
INFO | 21:10:22 13 [PowerFSM] Init the Device Information Service
INFO | 21:10:22 13 [PowerFSM] Init the Battery Service
INFO | 21:10:22 13 [PowerFSM] Init the Mesh bluetooth service
INFO | 21:10:22 13 [PowerFSM] Set up the advertising payload(s)
INFO | 21:10:22 13 [PowerFSM] Advertise
DEBUG | 21:10:22 13 [Screen] haveGlyphs=1
DEBUG | 21:10:22 13 [Screen] determineMode():
DEBUG | 21:10:22 13 [Screen] refresh=SKIPPED, reason=FRAME_MATCHED_PREVIOUS, frameFlags=0x3
INFO | 21:10:22 13 BLE Connected to LogicBloke S21 FE
DEBUG | 21:10:22 13 BluetoothStatus CONNECTED
INFO | 21:10:22 13 BLE connection secured
INFO | 21:10:23 14 [Screen] Done with boot screen
DEBUG | 21:10:23 14 [Screen] Show standard frames
DEBUG | 21:10:23 14 [Screen] Show 0 module frames
DEBUG | 21:10:23 14 [Screen] Total frame count: 83
DEBUG | 21:10:23 14 [Screen] Added modules. numframes: 0
DEBUG | 21:10:23 14 [Screen] Finished build frames. numframes: 3
DEBUG | 21:10:23 14 [Screen] determineMode():
DEBUG | 21:10:23 14 [Screen] refresh=SKIPPED, reason=FRAME_MATCHED_PREVIOUS, frameFlags=0x3
```

Issue seems to be happening at a different line of `main.cpp` now.

`ERROR | 21:08:28 9 NOTE! Record critical error 3 at src/main.cpp:1281`

Logicbloke, 2025-05-10:
[This is a comparison from 2.5.20 to 2.6.7](https://github.com/meshtastic/firmware/compare/v2.5.20.4c97351..v2.6.7.2d6181f#diff-49ed5df1ed6a707eff7c873df8a71754117f145522b003289a25a90c99d67a90)

Logicbloke, 2025-05-10:
It seems that `TCXO_OPTIONAL` is not defined for T-Echo, this makes it that [it will never enter the if-clause](https://github.com/meshtastic/firmware/blob/3a6fc668d8d799313fdcf068c5f201fe983f0566/src/main.cpp#L1094) where it tries to initialize SX1262 radio with XTAL.

Logicbloke, 2025-05-11:
I tried by removing
`#define NO_EXT_GPIO 1`

And adding
`#define TCXO_OPTIONAL`

Then it worked. 

I tried setting 
`#define SX126X_DIO3_TCXO_VOLTAGE 0.0`
while keeping 
`#define TCXO_OPTIONAL`

but it did not good.

Removing `SX126X_DIO3_TCXO_VOLTAGE` also caused some compilation errors.

skyboysea, 2025-06-25:
Fixed in 2.7.0.195b7cc.

closed 2025-06-25 by skyboysea