meshtastic/firmware #6300: [Feature Request]: Different default hop count for telemetry and maybe other message types

mikecarper, 2025-03-12:
### Platform

Cross-Platform

### Description

Set the default telemetry hop count to 1 by default. One can increase this value via the Meshtastic CLI if needed, and it can also be adjusted remotely while the node is in managed mode. Should help to reduce channel utilization across the network.

b8b8, 2025-03-12:
The newer firmware automatically scales back telemetry based on number of nodes in a mesh. Once the majority of people have upgraded, the telemetry congestion is no longer an issue. It is still important that the telemetry that does get out, goes out as far as the messages.

mikecarper, 2025-03-12:
yep I see this change
https://meshtastic.org/docs/overview/mesh-algo/#regular-broadcast-intervals

But none of this data is required for the network to function well; telemetry data is not as important as something like nodeinfo
https://meshtastic.org/docs/configuration/module/telemetry/

GUVWAF, 2025-03-12:
This is related to #6262, see also my comment there.
Although a limited hop limit might reduce channel utilization, if you're not interested in broadcasting it, it's much better to increase the broadcast interval.

mikecarper, 2025-03-12:
Marked as dup

closed 2025-03-12 by mikecarper