meshtastic/firmware #6301: [Feature Request]: Dynamic hop count for direct messages

mikecarper, 2025-03-12:
### Platform

Cross-Platform

### Description

User has a default hop count of 3 and successfully sends a DM to a BBS node. The BBS node should then reply using a dynamic setting that accounts for the hops already used and the remaining hops. For example, if the DM took 1 hop to reach the BBS node, the settings might offer options like Dynamic + 0, Dynamic + 1, or Dynamic + 2. With Dynamic + 0, the reply would use 1 hop; with Dynamic + 2, it would use 3 hops. This dynamic adjustment helps reduce overall channel utilization on the network.

GUVWAF, 2025-03-12:
For "official" request/reply packets used by the firmware, this is already implemented: 
https://github.com/meshtastic/firmware/blob/499ea56e3b825d45808d5f613c84196789ea1cd5/src/modules/RoutingModule.cpp#L64

There is no way for the firmware to know whether a text message was a reply to a previous message by a BBS. However, the BBS can set a hop limit per packet using the API itself.

GUVWAF, 2025-03-12:
Yes, for ACKs, but also other replies like traceroutes or NodeInfo. 

What I meant was that for each packet a client app (e.g. a BBS) sends to the node, they can set the hop limit. Only if it's not set, the node will use its configured value (when `wantAck` is set).

GUVWAF, 2025-03-12:
The `sendtext` is a command of the CLI. The actual API also has an option for the hop limit, which is configurable with the Python API's `sendData()` function: https://github.com/meshtastic/python/blob/master/meshtastic/mesh_interface.py#L480

mikecarper, 2025-03-12:
sendText() doesn't have that. Tomorrow I'll make an issue over in meshtastic/python to make ` hopLimit: Optional[int]=None,` available on all function calls
https://github.com/meshtastic/python/blob/master/meshtastic/mesh_interface.py#L406

closed 2025-03-13 by mikecarper