meshtastic/firmware #6304: [Bug]: Meshtasticd doesn't start. Assert Error

qkdxorjs1002, 2025-03-13:
### Category

Hardware Compatibility

### Hardware

Linux Native

### Firmware Version

2.6.1.9~obs7c3edde~

### Description

When I tried to set up and start meshtasticd on Raspberry Pi, I was experiencing the following error.
If you set DIO3_TCXO_VOLTAGE, an error appears. If you don't set it, you can't detect the module with -707 error.
The module rx/tx signal is tested with my testing code and RTL-SDR.

**Hardware**
Host: Raspberry Pi CM5 Lite
OS: Raspbian 12 Bookworm (6.12.x)
Module: E22-900M22S
Interface: SPI (on RPi CM5 SPI1-CE0)
Region: KR (921MHZ-923MHz)

**Configuration**
```yaml
Logging:
  LogLevel: debug
Lora:
  Module: sx1262
  DIO3_TCXO_VOLTAGE: 1.8
  SX126X_MAX_POWER: 22
  gpiochip: 4
  spidev: spidev1.0
#  CS: 18
  Reset: 14
  Busy: 15
```

### Relevant log output

```Shell
3월 13 10:44:02 uconsole systemd[1]: Started meshtasticd.service - Meshtastic Native Daemon.
 3월 13 10:44:02 uconsole meshtasticd[8834]: Portduino is starting, VFS root at /root/.portduino/default
 3월 13 10:44:02 uconsole meshtasticd[8834]: Set up Meshtastic on Portduino...
 3월 13 10:44:02 uconsole meshtasticd[8834]: Using /etc/meshtasticd/config.yaml as config file
 3월 13 10:44:02 uconsole meshtasticd[8834]: Also using "/etc/meshtasticd/config.d/lora-ebyte-e22-900m22s.yaml" as additional config file
 3월 13 10:44:02 uconsole meshtasticd[8834]: meshtasticd: pio/packages/framework-portduino/cores/portduino/PortduinoGPIO.cpp:48: GPIOPinIf* getGPIO(pin_size_t): Assertion `n < NUM_GPIOS' failed.
 3월 13 10:44:02 uconsole systemd[1]: meshtasticd.service: Main process exited, code=killed, status=6/ABRT
 3월 13 10:44:02 uconsole systemd[1]: meshtasticd.service: Failed with result 'signal'.
```

GPSFan, 2025-03-13:
I have had a similar problem with my LuckFox pico mini since 2.5.19. I wound up digging into RadioLib and patching 126x.cpp in modSetup to remove state = reset(), state = standby and their associated ASSERTs. This is because the findChip() code already does a reset and I believe that doing 2 resets so close together causes the SX1262 some problems. 
There are comments in the code indication that after a reset() the SX1262 doesn't respond to commands for a while and they supposedly take steps to address this. However in ther modSetup code the comment just before the reset says:
  // reset the module and verify startup

Yet the reset() invocation isn't reset(true) so it won't verify the SX1262 is ok before returning. 
I tried a reset(true and it didn't make any difference, but removing the reset() and standby() fixed the problem.

I don't know if this is the right solution, but it worked for me on each new meshtasticd release since.

qkdxorjs1002, 2025-03-13:
In SX126x constructor, parameters are -1 when value is not set.
I guessed that essential pin assertion this caused an error.
So I solved it by setting up an IRQ pin that was not actually assigned.

```
Logging:
  LogLevel: debug
Lora:
  Module: sx1262
  DIO2_AS_RF_SWITCH: true
  DIO3_TCXO_VOLTAGE: 1.8
  gpiochip: 4
  spidev: spidev1.0
  spiSpeed: 8000000
#  CS: 18
  IRQ: 17
  Reset: 14
  Busy: 15
```

GPSFan, 2025-03-13:
That spiSpeed looks awfully high...

markbirss, 2025-03-13:
> In SX126x constructor, parameters are -1 when value is not set. I guessed that essential pin assertion this caused an error. So I solved it by setting up an IRQ pin that was not actually assigned. [...]

Ok, does it show up in the debug with that value also?

DEBUG | 14:18:14 0 SX126xInterface(cs=0, irq=6, rst=2, busy=4)

[image]

qkdxorjs1002, 2025-03-14:
> Ok, does it show up in the debug with that value also?
>
> DEBUG | 14:18:14 0 SX126xInterface(cs=0, irq=6, rst=2, busy=4)
>
> [image]

```sh
  ~ ·················································································· with paragonnov@uconsole at  10:15:47
❯ sudo meshtasticd
Portduino is starting, VFS root at /root/.portduino/default
Set up Meshtastic on Portduino...
Using /etc/meshtasticd/config.yaml as config file
Also using "/etc/meshtasticd/config.d/lora-ebyte-e22-900m22s.yaml" as additional config file
MAC ADDRESS: 2C:CF:67:D8:9F:41
DEBUG | ??:??:?? 0 Upgrade time to quality NTP
DEBUG | 10:15:50 0 Read RTC time as 1741914950
INFO  | 10:15:50 0

//\ E S H T /\ S T / C

DEBUG | 10:15:50 0 Filesystem files:
DEBUG | 10:15:50 0  /. (0 Bytes)
DEBUG | 10:15:50 0  /prefs/nodes.proto (477 Bytes)
DEBUG | 10:15:50 0  /prefs/device.proto (102 Bytes)
DEBUG | 10:15:50 0  /prefs/module.proto (92 Bytes)
DEBUG | 10:15:50 0  /prefs/. (0 Bytes)
DEBUG | 10:15:50 0  /prefs/channels.proto (57 Bytes)
DEBUG | 10:15:50 0  /prefs/config.proto (172 Bytes)
DEBUG | 10:15:50 0  /prefs/.. (0 Bytes)
DEBUG | 10:15:50 0  /.. (0 Bytes)
INFO  | 10:15:50 0 No I2C device configured, Skip
INFO  | 10:15:50 0 No I2C devices found
INFO  | 10:15:50 0 S:B:37,2.6.1.1acc68bb
INFO  | 10:15:50 0 Init NodeDB
INFO  | 10:15:50 0 Load /prefs/nodes.proto
INFO  | 10:15:50 0 Loaded /prefs/nodes.proto successfully
INFO  | 10:15:50 0 Loaded saved nodedatabase version 24, with nodes count: 1
INFO  | 10:15:50 0 Load /prefs/device.proto
INFO  | 10:15:50 0 Loaded /prefs/device.proto successfully
INFO  | 10:15:50 0 Loaded saved devicestate version 24
INFO  | 10:15:50 0 Load /prefs/config.proto
INFO  | 10:15:50 0 Loaded /prefs/config.proto successfully
INFO  | 10:15:50 0 Loaded saved config version 24
INFO  | 10:15:50 0 Load /prefs/module.proto
INFO  | 10:15:50 0 Loaded /prefs/module.proto successfully
INFO  | 10:15:50 0 Loaded saved moduleConfig version 24
INFO  | 10:15:50 0 Load /prefs/channels.proto
INFO  | 10:15:50 0 Loaded /prefs/channels.proto successfully
INFO  | 10:15:50 0 Loaded saved channelFile version 24
/root/.portduino/default/prefs/uiconfig.proto does not exist
ERROR | 10:15:50 0 Could not open / read /prefs/uiconfig.proto
DEBUG | 10:15:50 0 cleanupMeshDB purged 0 entries
DEBUG | 10:15:50 0 Use nodenum 0x67d89f41
DEBUG | 10:15:50 0 Expand short PSK #1
INFO  | 10:15:50 0 Wanted region 7, using KR
DEBUG | 10:15:50 0 Coerce telemetry to min of 30 minutes on defaults
DEBUG | 10:15:50 0 Save to disk 0
DEBUG | 10:15:50 0 Use compiled/slipstreamed tzplaceholder
DEBUG | 01:15:50 0 Set Timezone to GMT0
DEBUG | 01:15:50 0 Read RTC time as 1741914950
DEBUG | 01:15:50 0 NeighborInfoModule is disabled
/root/.portduino/default/prefs/cannedConf.proto does not exist
ERROR | 01:15:50 0 Could not open / read /prefs/cannedConf.proto
INFO  | 01:15:50 0 CannedMessageModule is enabled
INFO  | 01:15:50 0 External Notification Module Disabled
DEBUG | 01:15:50 0 Activate sx1262 radio on SPI port /dev/spidev1.0
DEBUG | 01:15:50 0 SX126xInterface(cs=-1, irq=17, rst=14, busy=15)
DEBUG | 01:15:50 0 SX126X_DIO3_TCXO_VOLTAGE defined, using DIO3 as TCXO reference voltage at 1.800000 V
INFO  | 01:15:50 0 Start meshradio init
INFO  | 01:15:50 0 Radio freq=922.875, config.lora.frequency_offset=0.000
INFO  | 01:15:50 0 Set radio: region=KR, name=LongFast, config=0, ch=11, power=17
INFO  | 01:15:50 0 myRegion->freqStart -> myRegion->freqEnd: 920.000000 -> 923.000000 (3.000000 MHz)
INFO  | 01:15:50 0 numChannels: 12 x 250.000kHz
INFO  | 01:15:50 0 channel_num: 12
INFO  | 01:15:50 0 frequency: 922.875000
INFO  | 01:15:50 0 Slot time: 28 msec
INFO  | 01:15:50 0 Set radio: final power level=17
INFO  | 01:15:50 0 SX126x init result 0
INFO  | 01:15:50 0 Frequency set to 922.875000
INFO  | 01:15:50 0 Bandwidth set to 250.000000
INFO  | 01:15:50 0 Power output set to 17
DEBUG | 01:15:50 0 Current limit set to 140.000000
DEBUG | 01:15:50 0 Current limit set result 0
DEBUG | 01:15:50 0 Set DIO2 as not RF switch, result: 0
DEBUG | 01:15:50 0 Use MCU pin -1 as RXEN and pin -1 as TXEN to control RF switching
INFO  | 01:15:50 0 Set RX gain to boosted mode; result: 0
INFO  | 01:15:50 0 sx1262 init success
INFO  | 01:15:50 0 Use webserver port from yaml config 443
INFO  | 01:15:50 0 Webserver started
INFO  | 01:15:50 0 Web Server framework started on port: 443
INFO  | 01:15:50 0 Web Server root /usr/share/meshtasticd/web
INFO  | 01:15:50 0 API server listen on TCP port 4403
DEBUG | 01:15:50 0 LoRA bitrate = 118.394310 bytes / sec
INFO  | 01:15:50 0 PowerFSM init, USB power=1
DEBUG | 01:15:50 0 State: BOOT
```

Yes, there it is

```sh
DEBUG | 01:15:50 0 Activate sx1262 radio on SPI port /dev/spidev1.0
DEBUG | 01:15:50 0 SX126xInterface(cs=-1, irq=17, rst=14, busy=15)
DEBUG | 01:15:50 0 SX126X_DIO3_TCXO_VOLTAGE defined, using DIO3 as TCXO reference voltage at 1.800000 V
INFO  | 01:15:50 0 Start meshradio init
```

qkdxorjs1002, 2025-03-14:
But Tx/Rx doesn't seem to work. Tx works for handwritten python test code with the same settings.
Tx has no output in the SDR, and Rx pretends not to know the Heltec V3 node.
Below is the log when a message was sent to a channel.

```log
DEBUG | 01:37:59 107 [RadioIf] handleAPIv1ToRadio web -> radio
DEBUG | 01:37:59 107 [RadioIf] Received 79 bytes from PUT request
DEBUG | 01:37:59 107 [RadioIf] PACKET FROM PHONE (id=0x2c0339b0 fr=0x67d89f41 to=0xffffffff, WantAck=1, HopLim=0 Ch=0x0 Portnum=1 rxtime=1741916279)
DEBUG | 01:37:59 107 [RadioIf] handleReceived(USER) (id=0x2c0339b0 fr=0x00000000 to=0xffffffff, WantAck=1, HopLim=0 Ch=0x0 Portnum=1 rxtime=1741916279)
DEBUG | 01:37:59 107 [RadioIf] Module 'text' wantsPacket=1
INFO  | 01:37:59 107 [RadioIf] Received text msg from=0x0, id=0x2c0339b0, msg=txtestatxtestatxtestatxtestatxtestatxtestatxtesta
DEBUG | 01:37:59 107 [RadioIf] Module 'text' considered
DEBUG | 01:37:59 107 [RadioIf] Module 'StoreForward' wantsPacket=1
DEBUG | 01:37:59 107 [RadioIf] Module 'StoreForward' considered
DEBUG | 01:37:59 107 [RadioIf] Module 'ExternalNotificationModule' wantsPacket=1
INFO  | 01:37:59 107 [RadioIf] External Notification Module Disabled or muted
DEBUG | 01:37:59 107 [RadioIf] Module 'ExternalNotificationModule' considered
DEBUG | 01:37:59 107 [RadioIf] Module 'routing' wantsPacket=1
INFO  | 01:37:59 107 [RadioIf] Received routing from=0x0, id=0x2c0339b0, portnum=1, payloadlen=49
DEBUG | 01:37:59 107 [RadioIf] Routing sniffing (id=0x2c0339b0 fr=0x00000000 to=0xffffffff, WantAck=1, HopLim=0 Ch=0x0 Portnum=1 rxtime=1741916279)
DEBUG | 01:37:59 107 [RadioIf] Module 'routing' considered
DEBUG | 01:37:59 107 [RadioIf] Setting next retransmission in 7562 msecs:
DEBUG | 01:37:59 107 [RadioIf]  (id=0x2c0339b0 fr=0x00000000 to=0xffffffff, WantAck=1, HopLim=3 Ch=0x0 Portnum=1 rxtime=1741916279)
DEBUG | 01:37:59 107 [RadioIf] Add packet record fr=0x0, id=0x2c0339b0
DEBUG | 01:37:59 107 [RadioIf] Expand short PSK #1
DEBUG | 01:37:59 107 [RadioIf] Use AES128 key!
DEBUG | 01:37:59 107 [RadioIf] enqueue for send (id=0x2c0339b0 fr=0x67d89f41 to=0xffffffff, WantAck=0, HopLim=3 Ch=0x8 encrypted len=71 rxtime=1741916279 hopStart=3 relay=0x41 priority=100)
DEBUG | 01:37:59 107 [RadioIf] txGood=0,txRelay=0,rxGood=0,rxBad=0
DEBUG | 01:37:59 107 [RadioIf] end web->radio
DEBUG | 01:37:59 107 [RadioIf] FromRadio=STATE_SEND_PACKETS
```

madeofstown, 2025-03-19:
Instead of defining `DIO3_TCXO_VOLTAGE: 1.8` can you try `DIO3_TCXO_VOLTAGE: true`

Szetya, 2025-03-22:
In case SX1262 or SX1268 radio is defined and uses TCXO, it is necessary to define the macros SX126X_DIO3_TCXO_VOLTAGE (voltage) and TCXO_OPTIONAL in the variant.h file.

RADIOLIB_ERR_SPI_CMD_FAILED   (-707) : SX126x failed to execute SPI command. Often this means that the module is trying to use TCXO while XTAL is connected (or vice versa). Make sure your crystal setup (e.g. TCXO reference voltage) matches your hardware by setting "tcxoVoltage" to 0 when using XTAL module, or to appropriate value when using TCXO module.

``` cpp
#if defined(SX126X_DIO3_TCXO_VOLTAGE) && defined(TCXO_OPTIONAL)
    if ((!rIf) && (config.lora.region != meshtastic_Config_LoRaConfig_RegionCode_LORA_24)) {
        // try using the specified TCXO voltage
        auto *sxIf = new SX1268Interface(RadioLibHAL, SX126X_CS, SX126X_DIO1, SX126X_RESET, SX126X_BUSY);
        sxIf->setTCXOVoltage(SX126X_DIO3_TCXO_VOLTAGE);
        if (!sxIf->init()) {
            LOG_WARN("No SX1268 radio with TCXO, Vref %fV", SX126X_DIO3_TCXO_VOLTAGE);
            delete sxIf;
            rIf = NULL;
        } else {
            LOG_INFO("SX1268 init success, TCXO, Vref %fV", SX126X_DIO3_TCXO_VOLTAGE);
            rIf = sxIf;
            radioType = SX1268_RADIO;
        }
    }
#endif
    if ((!rIf) && (config.lora.region != meshtastic_Config_LoRaConfig_RegionCode_LORA_24)) {
        rIf = new SX1268Interface(RadioLibHAL, SX126X_CS, SX126X_DIO1, SX126X_RESET, SX126X_BUSY);
        if (!rIf->init()) {
            LOG_WARN("No SX1268 radio");
            delete rIf;
            rIf = NULL;
        } else {
            LOG_INFO("SX1268 init success");
            radioType = SX1268_RADIO;
        }
    }
```
@NomDeTom 
https://github.com/meshtastic/firmware/issues/6294

NomDeTom, 2025-03-22:
@Szetya tcxo-optional is only needed for designs that *could have* either type, not for one or the other. For Meshtasticd, I would expect the Xtal/TCXO status to be explicit, since it is so easy to set, and there is nothing to be gained by having it try twice at every startup.

Szetya, 2025-03-22:
> [@Szetya](https://github.com/Szetya) tcxo-optional is only needed for designs that _could have_ either type, not for one or the other. For Meshtasticd, I would expect the Xtal/TCXO status to be [...]

Only by using the two macros together can I bring the DIY Ebyte E22400M33S radio to life. If there is no TCXO_OPTIONAL then there is no `sxIf->setTCXOVoltage(SX126X_DIO3_TCXO_VOLTAGE);` and I get -707 error.

NomDeTom, 2025-03-22:
> Only by using the two macros together can I bring the DIY Ebyte E22400M33S radio to life. If there is no TCXO_OPTIONAL then there is no `sxIf->setTCXOVoltage(SX126X_DIO3_TCXO_VOLTAGE);` and I get -707 error.

I'm probably missing something. Is this for Linux native, or something else?

Szetya, 2025-03-23:
The above mentioned part is not linux. But I guess portduino is (I don't know linux native). And the same here. Screenshot from main.cpp:

[image]

closed 2025-05-15 by github-actions[bot]