meshtastic/firmware #6306: [Bug]: Transmit power is fixed to 17dBm

qkdxorjs1002, 2025-03-13:
### Category

Hardware Compatibility

### Hardware

Heltec V3

### Firmware Version

2.6.1.7

### Description

In Heltec V3.2, no matter what value the Transmit power setting is changed to, it is fixed at 17dBm.

The regional setting is Korea. 920-923MHz

### Relevant log output

```Shell

```

fifieldt, 2025-03-13:
looking at https://github.com/meshtastic/firmware/blob/499ea56e3b825d45808d5f613c84196789ea1cd5/src/mesh/RadioInterface.cpp#L78

It seems the power limit is specified as 0, whereas or should be a positive number.

fifieldt, 2025-03-13:
@qkdxorjs1002 do you happen to have a link to the power regulations in Korea? (Any language is OK)

qkdxorjs1002, 2025-03-13:
@fifieldt Thank you for comment.
This is an inquiry site about the latest laws of the Korean government.

https://www.law.go.kr/LSW/admRulLsInfoP.do?admRulId=53943&efYd=0

According to the Korean Radio Act, the regulation of the frequency band used in LoRa is as follows.
**Header - ` 제8조(RFID/USN 등의 무선설비) ① 917∼923.5 ㎒ 주파수대역의 전파를 사용하는 RFID용 무선설비의 기술기준은 다음 각 호와 같다. `**

[image]

**Transmit Power**
Original KR
```
4. 안테나절대이득을 포함한 복사전력은 10 mW 이하 (채널 1, 3, 4, 6, 7, 9, 10, 12, 13, 15, 16, 18번에서는 3 mW 이하)일 것. 다만, 수동형 RFID 판독기와 기록기의 경우 채널 2, 5, 8, 11, 14, 17에서 4 W이하, 채널 20부터 32까지는 200 mW 이하일 것 
```

ChatGPT EN Translation
```
4. The radiated power, including the antenna's absolute gain, shall not exceed 10 mW. However, for channels 1, 3, 4, 6, 7, 9, 10, 12, 13, 15, 16, and 18, it shall not exceed 3 mW.
In the case of passive RFID readers and writers, the power shall not exceed 4 W for channels 2, 5, 8, 11, 14, and 17, and 200 mW for channels 20 to 32.
```

qkdxorjs1002, 2025-03-13:
[image]

qkdxorjs1002, 2025-03-13:
LoraWAN Aliance's Regional Parameter

https://resources.lora-alliance.org/technical-specifications/rp002-1-0-4-regional-parameters

[image]

fifieldt, 2025-03-13:
Fixed by #6307

closed 2025-03-13 by fifieldt