meshtastic/firmware #6312: [Bug]: device-install.sh -p not targeting user specified port

ThatKalle, 2025-03-13:
### Category

Other

### Hardware

Heltec V3

### Firmware Version

2.6.1.7c3edde

### Description

From discord user wperry1: [link](https://discord.com/channels/867578229534359593/938197090935398451/1349518539521921126)

> Am I doing something wrong here? Ran the firmware-install with -p and the port but it installed to the Heltec in the other port. Not a big deal as I am just getting started and have my configs backed up but this could really mess someone up if they don't have the other device backed up.

### Relevant log output

```Shell
>./device-install.sh -p /dev/ttyUSB0 -f firmware-heltec-v3-2.6.1.7c3edde.bin --web
Trying to flash firmware-heltec-v3-2.6.1.7c3edde.bin, but first erasing and writing system information
esptool.py v4.8.1
Found 2 serial ports
Serial port /dev/ttyUSB1
```

ThatKalle, 2025-03-13:
The device-install/update scripts use `ESPTOOL_PORT` environment variable to target the user supplied port value, as per Esptool [documentation](https://docs.espressif.com/projects/esptool/en/latest/esp32/esptool/basic-options.html#serial-port-port-p).

I got myself 2x heltec v3s inbound to test with.

gjelsoe, 2025-03-15:
Try repleacing `ESPTOOL_PORT="$2"` with `export ESPTOOL_PORT="$2"`

PR #6341 (2025-03-18): device-install/update: fix esptool --port
This PR adds `--port INPUT` to the esptool cmd used during flashing. This way the script targets the port the user specified.

Testing on Windows, the previous solution of `SET ESPTOOL_PORT` worked.
`device-update.sh` did have `ESPTOOL_PORT` set via `export ESPTOOL_PORT`, but `device-install.sh` did not.

All 4 scripts have been adjusted to use the `--port INPUT` syntax.

---

_also did some housekeeping, moving log statements, fixing help and log messages_

fixes: #6312

---

I have tested the new version of `device-install.bat` on my Windows host suc [... cut]
closed 2025-03-18 by thebentern