meshtastic/firmware #6313: rxen, txen, and SX128X_POWER_EN support for ch341

jp-bennett, 2025-03-13:
### Platform

Linux Native

### Description

The CH341 support doesn't include rxen, txen, or SX128X_POWER_EN support, particularly for the sx128x chips. Our current native support is toggling those lines somewhat manually in SX128xInterface.cpp, using the portduino provided functions.

caveman99, 2025-03-23:
Are there even enough lines on the 341? Also is it worth the hassle? SX1262 and the LR series use their own IO lines to switch those toggles (apart from the power_en, which is always manual). This would only ever be relevant to a SX1280, maybe even with PA.

markbirss, 2025-03-23:
Yes, there are actually upto 15 gpios in total, just first 7 is bi-directional

Im now using D1,D2 for RXEN and TXEN

Ok, so, i did some more testing seems it might actually be working now, must just confirm rf power test

I get upto 2w draw from USB port, thus sx1280 is using a good amount of current (together sx1280 and ch341 using 0.606 A)

CH341 at idle using about 125 mA from USB 2.0 port but goes upto 400 mA which is upto 606 mA ouput on 3v3 by LDO

and can even feel the module producing heat

I had to modify RadioLibInterface to set power to 4 and although i get Unbound messages it ok

closed 2025-05-15 by github-actions[bot]