meshtastic/firmware #6315: [Bug]: Meshtastic 2.6.1 + CardKB + Canned Message + DM = Primary Channel

Xaositek, 2025-03-14:
### Category

Hardware Compatibility

### Hardware

Rak4631

### Firmware Version

2.6.1.7c3edde

### Description

I updated my Meshenger (RAK4631 + CardKB + OLED Screen) and as testing canned messages. I found that with Meshtastic 2.6.1, if you send a DM with Canned Messages, it goes to the primary channel. Verified that other DMs are working as expected.

This is a photo of the message I'm about to send to the mesh.
[image]

Here is the message on LongFast (Primary Channel) that was sent by Meshenger.
[image]

### Relevant log output

```Shell
Below is the entire long but this one seems relevant:
INFO  | 22:21:52 952 [CannedMessage] Send message id=2093648260, dest=ffffffff, msg=/wx nb

Connected to radio
DEBUG | 22:21:35 936 [cardKB] Show standard frames
DEBUG | 22:21:35 936 [cardKB] canned wants a UI Frame
DEBUG | 22:21:35 936 [cardKB] Show 1 module frames
DEBUG | 22:21:35 936 [cardKB] Total frame count: 84
DEBUG | 22:21:35 936 [cardKB] Added modules.  numframes: 1
DEBUG | 22:21:35 936 [cardKB] Finished build frames. numframes: 8
DEBUG | 22:21:40 941 [cardKB] First touch (0):Good day all
DEBUG | 22:21:40 941 [cardKB] Show standard frames
DEBUG | 22:21:40 941 [cardKB] canned wants a UI Frame
DEBUG | 22:21:40 941 [cardKB] Show 1 module frames
DEBUG | 22:21:40 941 [cardKB] Total frame count: 84
DEBUG | 22:21:40 941 [cardKB] Added modules.  numframes: 1
DEBUG | 22:21:40 941 [cardKB] Finished build frames. numframes: 8
DEBUG | 22:21:42 943 [cardKB] MOVE UP (8):/forecast nb
DEBUG | 22:21:45 946 [Power] Battery: usbPower=1, isCharging=1, batMv=3954, batPct=76
DEBUG | 22:21:49 949 [cardKB] MOVE UP (7):/wx nb
DEBUG | 22:21:50 951 [cardKB] MOVE UP (6):/ping
DEBUG | 22:21:51 952 [cardKB] MOVE DOWN (7):/wx nb
DEBUG | 22:21:52 952 [CannedMessage] Partially randomized packet id 2093648260
INFO  | 22:21:52 952 [CannedMessage] Send message id=2093648260, dest=ffffffff, msg=/wx nb
DEBUG | 22:21:52 952 [CannedMessage] Update DB node 0x2f224e91, rx_time=1741922512
DEBUG | 22:21:52 952 [CannedMessage] handleReceived(LOCAL) (id=0x7cca8984 fr=0x2f224e91 to=0xffffffff, WantAck=1, HopLim=7 Ch=0x0 Portnum=1 rxtime=1741922512)
DEBUG | 22:21:52 952 [CannedMessage] No modules interested in portnum=1, src=LOCAL
DEBUG | 22:21:52 952 [CannedMessage] Setting next retransmission in 6924 msecs: 
DEBUG | 22:21:52 952 [CannedMessage]  (id=0x7cca8984 fr=0x2f224e91 to=0xffffffff, WantAck=1, HopLim=7 Ch=0x0 Portnum=1 rxtime=1741922512)
DEBUG | 22:21:52 952 [CannedMessage] Add packet record fr=0x2f224e91, id=0x7cca8984
DEBUG | 22:21:52 952 [CannedMessage] Expand short PSK #1
DEBUG | 22:21:52 952 [CannedMessage] Use AES128 key!
DEBUG | 22:21:52 953 [CannedMessage] enqueue for send (id=0x7cca8984 fr=0x2f224e91 to=0xffffffff, WantAck=0, HopLim=7 Ch=0x8 encrypted len=28 rxtime=1741922512 hopStart=7 relay=0x91 priority=100)
DEBUG | 22:21:52 953 [CannedMessage] txGood=16,txRelay=8,rxGood=125,rxBad=6
DEBUG | 22:21:52 953 [CannedMessage] Use channel 0 (hash 0x8)
DEBUG | 22:21:52 953 [CannedMessage] Expand short PSK #1
DEBUG | 22:21:52 953 [CannedMessage] Use AES128 key!
DEBUG | 22:21:52 953 [CannedMessage] decoded message (id=0x7cca8984 fr=0x2f224e91 to=0xffffffff, WantAck=0, HopLim=7 Ch=0x0 Portnum=1 rxtime=1741922512 hopStart=7 relay=0x91 priority=100)
DEBUG | 22:21:52 953 [CannedMessage] Show standard frames
DEBUG | 22:21:52 953 [CannedMessage] canned wants a UI Frame
DEBUG | 22:21:52 953 [CannedMessage] Show 1 module frames
DEBUG | 22:21:52 953 [CannedMessage] Total frame count: 84
DEBUG | 22:21:52 953 [CannedMessage] Added modules.  numframes: 1
DEBUG | 22:21:52 953 [CannedMessage] Finished build frames. numframes: 8
INFO  | 22:21:52 953 Tell client we have new packets 68
[... 86 lines cut ...]
INFO  | 22:21:56 957 [Router] externalNotificationModule - Notification Module (Buzzer)
DEBUG | 22:21:56 957 [Router] Module 'ExternalNotificationModule' considered
DEBUG | 22:21:56 957 [Router] Module 'routing' wantsPacket=1
INFO  | 22:21:56 957 [Router] Received routing from=0xeb6171dd, id=0xe9691c72, portnum=1, payloadlen=106
DEBUG | 22:21:56 957 [Router] Routing sniffing (id=0xe9691c72 fr=0xeb6171dd to=0x2f224e91, WantAck=0, HopLim=6 Ch=0x0 Portnum=1 PKI rxtime=1741922516 rxSNR=7.75 rxRSSI=-66 via MQTT hopStar
DEBUG | 22:21:56 957 [Router] Delivering rx packet (id=0xe9691c72 fr=0xeb6171dd to=0x2f224e91, WantAck=0, HopLim=6 Ch=0x0 Portnum=1 PKI rxtime=1741922516 rxSNR=7.75 rxRSSI=-66 via MQTT hop
DEBUG | 22:21:56 957 [Router] Update DB node 0xeb6171dd, rx_time=1741922516
DEBUG | 22:21:56 957 [Router] Forwarding to phone (id=0xe9691c72 fr=0xeb6171dd to=0x2f224e91, WantAck=0, HopLim=6 Ch=0x0 Portnum=1 PKI rxtime=1741922516 rxSNR=7.75 rxRSSI=-66 via MQTT hopS
DEBUG | 22:21:56 957 [Router] Module 'routing' considered
INFO  | 22:21:56 957 Tell client we have new packets 70
INFO  | 22:21:56 957 BLE notify fromNum
DEBUG | 22:21:56 957 [ExternalNotification] EXTERNAL 2 1000 compared to 957400
DEBUG | 22:21:56 957 FromRadio=STATE_SEND_PACKETS
DEBUG | 22:21:56 957 phone downloaded packet (id=0xe9691c72 fr=0xeb6171dd to=0x2f224e91, WantAck=0, HopLim=6 Ch=0x0 Portnum=1 PKI rxtime=1741922516 rxSNR=7.75 rxRSSI=-66 via MQTT 
DEBUG | 22:21:57 957 [RadioIf] Lora RX (id=0x7cca8984 fr=0x2f224e91 to=0xffffffff, WantAck=0, HopLim=6 Ch=0x8 encrypted len=28 rxSNR=6.75 rxRSSI=-34 via MQTT hopStart=7)
DEBUG | 22:21:57 957 [RadioIf] Packet RX: 444ms
DEBUG | 22:21:57 957 [Router] Rx someone rebroadcasting for us (id=0x7cca8984 fr=0x2f224e91 to=0xffffffff, WantAck=0, HopLim=6 Ch=0x8 encrypted len=28 rxSNR=6.75 rxRSSI=-34 via MQTT hopSta
DEBUG | 22:21:57 957 [Router] Didn't find pending packet
DEBUG | 22:21:57 957 [Router] Found existing packet record for fr=0x2f224e91,to=0xffffffff,id=0x7cca8984
DEBUG | 22:21:57 957 [Router] Add packet record fr=0x2f224e91, id=0x7cca8984
DEBUG | 22:21:57 957 [Router] Ignore dupe incoming msg (id=0x7cca8984 fr=0x2f224e91 to=0xffffffff, WantAck=0, HopLim=6 Ch=0x8 encrypted len=28 rxSNR=6.75 rxRSSI=-34 via MQTT hopStart=7)
DEBUG | 22:21:57 957 [Router] cancelSending id=0x7cca8984, removed=0
DEBUG | 22:21:57 957 [Router] Incoming msg was filtered from 0x2f224e91
DEBUG | 22:21:57 958 [ExternalNotification] EXTERNAL 2 958401 compared to 958411
```

linker3000, 2025-03-30:
If you need to send a simple message from a sensor and the strings come from the sensor, if there was an option to send the messages via the TEXTBUF serial function, this suggestion would support the need:

https://github.com/meshtastic/firmware/discussions/6461

RCGV1, 2025-04-01:
https://github.com/meshtastic/firmware/pull/6475

closed 2025-04-01 by thebentern

Xaositek, 2025-04-02:
Great work @RCGV1 - I just verified that everything works fantastic!

commit 64484912 (2025-04-01): Fixes #6315 (#6475)

* Fixed Canned Messages send to non broadcast

* Small fix

* Fix formatting for singular canned message

* Trunk fmt

---------