meshtastic/firmware #6324: [Bug]: T-Deck Pro Lilygo BT Pairing failure when wifi on

Slyke, 2025-03-15:
### Category

BLE

### Hardware

T-Deck

### Firmware Version

2.6.1.7c3edde

### Description

Hello

I installed the latest version of Meshtastic with MUI on 2 T-Deck Pros. I'm aware that only wifi or BT can be active at a time.

When rebooting into BT mode with Wifi turned off, I can pair with them with the Meshtastic app.

If I reboot back into MUI mode, then turn on wifi (reboots), then try to either connect with an IP address or BT in any mode (MUI or BT pairing) the device doesn't respond.

Essentially turning on wifi prevents the device from connecting to anything in any mode.

Turning wifi back off (and rebooting) allows pairing when the device is in BT pairing mode.

It seems that the wifi chip becomes active and prevents BT from working, even though wifi also doesn't work.

Potential solutions:
* When in BT mode, when pairing number is on the screen, allow switching between wifi and BT
* Turn wifi off when entering BT mode, and back on (if it was initially on) when exiting pairing mode
* Allow wifi to be optionally turned off in BT pairing mode
* Allow rebooting into MUI mode, BT pairing mode, or WUI (WebUI) mode, if there's enough space to put both MUI and WUI on the device

### Relevant log output

```Shell

```

garthvh, 2025-03-15:
The MUI, WiFi and BLE all use the same client api, only 1 can be used at a time.

caveman99, 2025-03-15:
connect via IP uses two endpoints on the webserver, which is switched off on the T-Deck with MUI. The TCP-API endpoint is disabled as well, for the reason @garthvh  mentioned. So if you switch on wifi, this is only useful as a client device, connecting to an MQTT server.

closed 2025-03-15 by caveman99

Slyke, 2025-03-15:
@garthvh 

> The MUI, WiFi and BLE all use the same client api, only 1 can be used at a time.

Yeah, I'm aware. My issue is that we can enter BT pairing mode and it will never pair with anything because wifi is on, and there's no indication that wifi is on, and that pairing will fail. It even has a pairing number on screen.

@caveman99 

> connect via IP uses two endpoints on the webserver, which is switched off on the T-Deck with MUI. The TCP-API endpoint is disabled as well, for the reason @garthvh mentioned. So if you switch on wifi, this is only useful as a client device, connecting to an MQTT server.

Shouldn't the ability to enter BT mode be disabled in that situation?