meshtastic/firmware #6340: [Bug]:

willp618, 2025-03-17:
### Category

WiFi

### Hardware

Heltec V3

### Firmware Version

2.5.20.4c97351 beta

### Description

MQTT in a connect disconnect loop 

mqtt seems to get stuck in a connect disconnect loop if not connected to a phone or tablet via wifi once connected to phone or tablet via ip it will connect to the mqtt server 
ive tried resetting the node updating to 2.6.1 degrading to 2.5.15 same issue client proxy is off everything was working till i updated a few days ago  
 
i also tried a seeed xiao running 2.6.1 and it also does the same 

### Relevant log output

```Shell
DEBUG | 14:09:55 1176 [Router] Incoming msg was filtered from 0xda64a898

[1176389][D][WiFiClient.cpp:536] connected(): Disconnected: RES: 0, ERR: 128

INFO | 14:09:55 1176 [mqtt] Use non-TLS-encrypted session

INFO | 14:09:55 1176 [mqtt] Connect directly to MQTT server mqtt.meshtastic.liamcottle.net, port: 1883, username: uplink, password: uplink

INFO | 14:09:55 1177 [mqtt] MQTT connected

DEBUG | 14:09:57 1178 [RadioIf] Lora RX (id=0xbd18b73c fr=0xa0 to=0x98, WantAck=0, HopLim=2 Ch=0x8 encrypted rxSNR=-14 rxRSSI=-120 hopStart=3)

DEBUG | 14:09:57 1178 [RadioIf] Packet RX: 665ms

DEBUG | 14:09:57 1178 [Router] Add packet record (id=0xbd18b73c fr=0xa0 to=0x98, WantAck=0, HopLim=2 Ch=0x8 encrypted rxSNR=-14 rxRSSI=-120 hopStart=3)

DEBUG | 14:09:57 1178 [Router] Use channel 0 (hash 0x8)

DEBUG | 14:09:57 1178 [Router] Expand short PSK #1

DEBUG | 14:09:57 1178 [Router] Use AES128 key!

DEBUG | 14:09:57 1178 [Router] decoded message (id=0xbd18b73c fr=0xa0 to=0x98, WantAck=0, HopLim=2 Ch=0x0 Portnum=4 WANTRESP rxtime=1742220597 rxSNR=-14 rxRSSI=-120 hopStart=3)

DEBUG | 14:09:57 1178 [Router] handleReceived(REMOTE) (id=0xbd18b73c fr=0xa0 to=0x98, WantAck=0, HopLim=2 Ch=0x0 Portnum=4 WANTRESP rxtime=1742220597 rxSNR=-14 rxRSSI=-120 hopStart=3)

DEBUG | 14:09:57 1178 [Router] Module 'nodeinfo' wantsPacket=1

INFO | 14:09:57 1178 [Router] Received nodeinfo from=0xda5c81a0, id=0xbd18b73c, portnum=4, payloadlen=36

DEBUG | 14:09:57 1178 [Router] old user Hanbury/Hnbr, channel=0

DEBUG | 14:09:57 1178 [Router] Update changed=0 user Hanbury/Hnbr, channel=0

DEBUG | 14:09:57 1178 [Router] Module 'nodeinfo' considered

DEBUG | 14:09:57 1178 [Router] Module 'neighborinfo' wantsPacket=1

INFO | 14:09:57 1178 [Router] Received neighborinfo from=0xda5c81a0, id=0xbd18b73c, portnum=4, payloadlen=36

DEBUG | 14:09:57 1178 [Router] Module 'neighborinfo' considered

DEBUG | 14:09:57 1178 [Router] Module 'routing' wantsPacket=1

INFO | 14:09:57 1178 [Router] Received routing from=0xda5c81a0, id=0xbd18b73c, portnum=4, payloadlen=36

DEBUG | 14:09:57 1178 [Router] Routing sniffing (id=0xbd18b73c fr=0xa0 to=0x98, WantAck=0, HopLim=2 Ch=0x0 Portnum=4 WANTRESP rxtime=1742220597 rxSNR=-14 rxRSSI=-120 hopStart=3)

INFO | 14:09:57 1178 [Router] Rebroadcast received floodmsg

DEBUG | 14:09:57 1179 [Router] Expand short PSK #1

DEBUG | 14:09:57 1179 [Router] Use AES128 key!

DEBUG | 14:09:57 1179 [Router] enqueue for send (id=0xbd18b73c fr=0xa0 to=0x98, WantAck=0, HopLim=1 Ch=0x8 encrypted rxtime=1742220597 rxSNR=-14 rxRSSI=-120 hopStart=3 priority=70)

DEBUG | 14:09:57 1179 [Router] txGood=61,txRelay=54,rxGood=211,rxBad=58

DEBUG | 14:09:57 1179 [Router] rx_snr found. hop_limit:1 rx_snr:-14.000000

DEBUG | 14:09:57 1179 [Router] rx_snr found in packet. Setting tx delay:1309

DEBUG | 14:09:57 1179 [Router] Module 'routing' considered

INFO | 14:09:57 1179 [Router] MQTT onSend - Not forwarding packet due to DontMqttMeBro flag

[1179327][D][WiFiClient.cpp:536] connected(): Disconnected: RES: 0, ERR: 128

INFO | 14:09:58 1179 [mqtt] Use non-TLS-encrypted session

INFO | 14:09:58 1179 [mqtt] Connect directly to MQTT server mqtt.meshtastic.liamcottle.net, port: 1883, username: uplink, password: uplink

INFO | 14:09:58 1180 [mqtt] MQTT connected

DEBUG | 14:09:59 1180 [RadioIf] Started Tx (id=0xbd18b73c fr=0xa0 to=0x98, WantAck=0, HopLim=1 Ch=0x8 encrypted rxtime=1742220597 rxSNR=-14 rxRSSI=-120 hopStart=3 priority=70)

DEBUG | 14:09:59 1180 [RadioIf] Packet TX: 665ms

DEBUG | 14:09:59 1181 [RadioIf] Completed sending (id=0xbd18b73c fr=0xa0 to=0x98, WantAck=0, HopLim=1 Ch=0x8 encrypted rxtime=1742220597 rxSNR=-14 rxRSSI=-120 hopStart=3 priority=70)

[1182193][D][WiFiClient.cpp:536] connected(): Disconnected: RES: 0, ERR: 128

INFO | 14:10:01 1182 [mqtt] Use non-TLS-encrypted session

INFO | 14:10:01 1182 [mqtt] Connect directly to MQTT server mqtt.meshtastic.liamcottle.net, port: 1883, username: uplink, password: uplink

INFO | 14:10:01 1182 [mqtt] MQTT connected

DEBUG | 14:10:01 1183 [RadioIf] Lora RX (id=0xbd18b73c fr=0xa0 to=0x98, WantAck=0, HopLim=0 Ch=0x8 encrypted rxSNR=-15.5 rxRSSI=-124 hopStart=3)

DEBUG | 14:10:02 1183 [RadioIf] Packet RX: 665ms

DEBUG | 14:10:02 1183 [Router] Found existing packet record for fr=0xda5c81a0,to=0xda64a898,id=0xbd18b73c

DEBUG | 14:10:02 1183 [Router] Add packet record (id=0xbd18b73c fr=0xa0 to=0x98, WantAck=0, HopLim=0 Ch=0x8 encrypted rxSNR=-15.5 rxRSSI=-124 hopStart=3)
```

fifieldt, 2025-03-21:
Thanks for the report and the logs!

fifieldt, 2025-03-21:
This seems related: https://github.com/espressif/arduino-esp32/issues/1921

thebentern, 2025-03-21:
Hi, would you be willing to try an experimental binary when this build completes?
https://github.com/meshtastic/firmware/pull/6363

A while back, someone made a change to increase the stability of the wifi connection for ESP32-C3 devices by using static buffers mode. I'd like to see if unlocking that for other ESP32 archs causes more stability there as well.

willp618, 2025-03-21:
ill give a go

PR #6363 (2025-03-29): Try-fix ESP32 wifi disconnects
Hopefully fixes #6340
closed 2025-05-30 by github-actions[bot]