meshtastic/firmware #6361: [Feature Request]: Disable amateur radio licensed mode on non-amateur bands

oherrala, 2025-03-21:
### Platform

Cross-Platform

### Description

I've seen some amateur radio operators turning on the "licensed operator" mode on EU 868 MHz band. That band is not radio amateur band.

This can be harmful to the operation of the whole network. For example these changes affect network in this scenario and there are probably more:

 * #6068 
 * #6047

Can this "licensed operator" mode be disabled on non-amateur radio bands (e.g. EU_868)?

madeofstown, 2025-03-21:
Excuse my ignorance but is it not the responsibility of the licensed operator to ensure they are broadcasting legally? 
But honestly, I think that there should be a separate firmware to flash if you are licensed. Maybe even have the download locked until the user puts in a valid callsign.

thebentern, 2025-03-21:
> Honestly, I think that there should be a separate firmware to flash if you are licensed. Maybe even have the download locked until the user puts in a valid callsign.

I think we can put some more common-sense safeguards in place without throwing out a whole separate curated firmware, which frankly is a lot of extra distribution overhead I'm not very interested in at the moment. 

But to your point about the responsibility of the operator, it's the same thing as someone going out and picking up a Baofeng radio and operating it on ham bands unlicensed. Nothing prevents you from doing that and operating illegally, and there are always ways around the safeguards. What I want to avoid is giving users easy ways to inadvertently violate the regulations

oherrala, 2025-03-21:
> What I want to avoid is giving users easy ways to inadvertently violate the regulations

Thanks @thebentern this is exactly what I wanted to hear. I'm happy 👍 

> Excuse my ignorance but is it not the responsibility of the licensed operator to ensure they are broadcasting legally?

@madeofstown I've seen Meshtastic enthusiastics (that don't have radio amateur license) tell ham operators to turn on the licensed mode because they are hams. Yes, the responsibility is by the radio operator, but new people getting into new cool tech can be confusing and misguided advice doesn't help. Meshtastic is one of those new cool radio techs that's easily picked up by people who are not trained on radio regulation, etc. but are really eager to share advice.

The iOS App mentions when turning on the Licensed Operator mode that the duty cycle restriction is removed on EU_868 band. This makes the firmware break [ETSI EN300.220 V3.2.1 (2018-06)](https://www.etsi.org/deliver/etsi_en/300200_300299/30022002/03.02.01_60/en_30022002v030201p.pdf) standard and is thus in violation to EU [Directive 2014/53/EU](https://eur-lex.europa.eu/eli/dir/2014/53/oj) and via that it is probably illegal in most member countries.

SimbimChimbetov, 2025-03-21:
in my region are many new to this and know zero about regulations.

I saw some messages like "set hop to 7" or "turn on ignore duty cycle", "set node as router_client" as a responses to questions why they can't hear their friends.

In the past I tried to explain and optimize our mesh but now I gave up because no one cared.
The neighbor city is now full of routers, turned off duty cycles on 868 EU