meshtastic/firmware #6368: [Bug]: Enabling the Alert Message/Bell LED settings also enables the Buzzer, even when the Alert Message/Bell Buzzer is switched off.

Supermath101, 2025-03-21:
### Category

Other

### Hardware

Seeed Card Tracker T1000-E

### Firmware Version

2.6.1.7c3edde

### Description

With the External Notification module enabled, if I enabled the LED alert settings, it also enables the Buzzer, even when the Buzzer alert setting is set to disabled. This bug occurs for both the Alert message LED/buzzer and Alert bell LED/buzzer setting pairs.

### Relevant log output

```Shell

```

fifieldt, 2025-03-21:
Looks like the code may just check if the buzzer GPIO is defined rather than if it's enabled. Easy fix, will test.

`
        // now let the PWM buzzer play
        if (moduleConfig.external_notification.use_pwm && config.device.buzzer_gpio) {
            if (rtttl::isPlaying()) {
                rtttl::play();
            } else if (isNagging && (nagCycleCutoff >= millis())) {
                // start the song again if we have time left
                rtttl::begin(config.device.buzzer_gpio, rtttlConfig.ringtone);
            }
        }
`

fifieldt, 2025-03-21:
hmm, use_pwm is probably the right value.

defaults:
```
 690 #if defined(PIN_BUZZER)
 691     moduleConfig.external_notification.enabled = true;
 692     moduleConfig.external_notification.output_buzzer = PIN_BUZZER;
 693     moduleConfig.external_notification.use_pwm = true;
 694     moduleConfig.external_notification.alert_message_buzzer = true;
 695     moduleConfig.external_notification.nag_timeout = 60;
 696 #endif

fifieldt, 2025-03-21:
With Alert message LED only, no buzzer sounds (2.6.2)

[image]

fifieldt, 2025-03-21:
With both Alert Message LED and Alert Bell LED enabled, no buzzer sounds. 

[image]

fifieldt, 2025-03-21:
Unfortunately, I can't reproduce this - are you able to export your settings and upload, or take a screenshot of the external notification page?

and, just to confirm - you're not talking about the startup/shutdown buzzer?

fifieldt, 2025-03-22:
Apologies, I just realised I was running 2.5.22, not 2.6.2

Supermath101, 2025-03-22:
> Unfortunately, I can't reproduce this - are you able to export your settings and upload, or take a screenshot of the external notification page?
> 
> and, just to confirm - you're not talking about the startup/shutdown buzzer?

I'm *not* talking about the startup/shutdown buzzer. Here's my config:
```yml
# start of Meshtastic configure yaml
channel_url: REDACTED
config:
  bluetooth:
    enabled: true
    fixedPin: REDACTED
    mode: FIXED_PIN
  device:
    buzzerGpio: 25
    disableTripleClick: true
    ledHeartbeatDisabled: true
    nodeInfoBroadcastSecs: 10800
    role: CLIENT_MUTE
    tzdef: EST5EDT,M3.2.0,M11.1.0
  display:
    screenOnSecs: 600
    units: IMPERIAL
  lora:
    hopLimit: 5
    ignoreMqtt: true
    region: US
    sx126xRxBoostedGain: true
    txEnabled: true
    txPower: 30
    usePreset: true
  network:
    ntpServer: meshtastic.pool.ntp.org
  position:
    broadcastSmartMinimumDistance: 100
    broadcastSmartMinimumIntervalSecs: 30
    gpsEnGpio: 43
    gpsMode: ENABLED
    gpsUpdateInterval: 15
    positionBroadcastSecs: 900
    positionBroadcastSmartEnabled: true
    positionFlags: 811
  power:
    lsSecs: 300
    minWakeSecs: 10
    sdsSecs: 4294967295
    waitBluetoothSecs: 60
  security:
    privateKey: REDACTED
    publicKey: base64:n71jei/uNpIqJdRZ4ZF5fVdjUOJ8adaFjQKrqOiZfhY=
    serialEnabled: true
location:
  alt: REDACTED
  lat: REDACTED
  lon: REDACTED
module_config:
  ambientLighting:
    blue: 125
    current: 10
    green: 161
    red: 205
  detectionSensor:
    detectionTriggerType: LOGIC_HIGH
    minimumBroadcastSecs: 45
  externalNotification:
    active: true
    alertMessage: true
    enabled: true
    nagTimeout: 1
    output: 24
    outputBuzzer: 25
    usePwm: true
  mqtt:
    address: mqtt.meshtastic.org
    encryptionEnabled: true
    password: large4cats
    root: msh/US
    username: meshdev
  rangeTest:
    enabled: true
owner: Supermath101 (a17d)
owner_short: a17d
```

fifieldt, 2025-03-22:
Just flashed 2.6.2 and confirmed the same results as before. Taking a look at the config!

fifieldt, 2025-03-22:
Interestingly, on 2.6 on the android app, this toggle that was present in 2.5 no longer appears!

Supermath101, 2025-03-22:
> > usePwm: true
> 
> ^ this is why the buzzer is going off, this poorly named field enables the PWM Buzzer

What if I only want the Buzzer enabled for Bell messages?

fifieldt, 2025-03-22:
Let me check the code!

fifieldt, 2025-03-22:
:( I want to rewrite this code.

garthvh, 2025-03-23:
@caveman99 added this and it is required for the most common buzzers to work (rak and t1000e)

caveman99, 2025-03-23:
Can you check if this is also happening for devices WITH buzzer but NOT being T1000e? I seem to recall the T1000e is handling the buzzing differently.

B1acks1te, 2025-05-13:
Can confirm that I am having this very issue on x2 T1000-E devices flashed to 2.5.20.4c97351 Beta as well as 2.6.4.b89355f Beta

Testing line up: 

TEST 1
Alert Message LED = Off
Alert Message Buzzer = Off

Alert Bell LED = Off
Alert Bell Buzzer = Off

Outcome = No buzzer, No LED

TEST 2
Alert Message LED = On
Alert Message Buzzer = Off

Alert Bell LED = Off
Alert Bell Buzzer = Off

Outcome = Buzzer and LED

TEST 3
Alert Message LED = Off
Alert Message Buzzer = Off

Alert Bell LED = On
Alert Bell Buzzer = Off

Outcome = Buzzer and LED

The outcome expected is that only get buzzer when alert bell notification comes through and not for all messages. Only way to get LED and no buzzer currently for me is to disable the PWM Buzzer option

B1acks1te, 2025-05-13:
> Can confirm that I am having this very issue on x2 T1000-E devices flashed to 2.5.20.4c97351 Beta as well as 2.6.4.b89355f Beta
>
> Testing line up:
>
> TEST 1 Alert Message LED = Off Alert Message Buzzer [...]

Further to this after doing some exhaustive testing with another user we came to the conclusion that the NAG TIMEOUT breaks the feature

When the NAG is set to anything other than 0 you will get buzzer on ALL messages but set to 0 it will work as you expect. Hopefully this is any easy patch

va3jme, 2025-07-30:
Experiencing same issue when configured node via iOS, but not when configured via Android. Android has a few extra options it seems