meshtastic/firmware #6369: [Bug]: Periodic NodeInfo Broadcast Interval Not Scaled ?

MeshTrafficObserver, 2025-03-21:
### Category

Other

### Hardware

Heltec V3

### Firmware Version

2.6.1

### Description

Hello,

I’ve been analyzing my NodeInfo transmissions (broadcast to address 4294967295). I’m running a CLIENT in LONG_FAST mode on a Heltec v3, and I noticed that the periodic NodeInfo broadcasts do not seem to be scaled based on the total number of nodes in the network.

In **NodeInfoModule.cpp** within the function **NodeInfoModule::runOnce()**, the broadcast interval is determined by the call:

Default::getConfiguredOrDefaultMs(config.device.node_info_broadcast_secs, default_node_info_broadcast_secs);

If NodeInfo should be scaled should this not be:

Default::getConfiguredOrDefault**MsScaled**(config.device.node_info_broadcast_secs, default_node_info_broadcast_secs, numOnlineNodes);

Or are only Position, Telemetry etc. transmission supposed to be scaled?

Anyway, the real issue is the Node-DB overflow and the 5 minutes check applied. In our huge network we have about 12% periodic NodeInfo transmission versus 88 % probably coming for Node-DB overflows. Many node firing NodeInfos every 6 – 8 Minutes with up to 100 Nodes online this gets quite a lot. Any chance to scale the 5 minutes check too ?

Thank you

### Relevant log output

```Shell

```

b8b8, 2025-03-21:
How many nodes in your network? Might be too busy for longfast and need a faster preset.

MeshTrafficObserver, 2025-03-21:
Within a 6 day period in Feb25 there have been 441 individual nodes seen. 
The count of active nodes per interval and the role distribution looked as follow:

[image]
[image]

Thanks for the reference b8b8. The increase of the Node-DB size in 2.6.2 will certainly help.

I just think user believe that if they set the NodeInfo intervall to 3h, the NodeInfo is only sent out every 3 hours.
The reality however looks (at least in a big network) very different.

Below are two additional graphs showing the same 3 day time frame:

The graph below shows all NodeInfo transmissions which should mainly be periodic transmissions (the ones which are sent to Broadcast NodeNum 4294967295)
[image]

This graph shows all NodeInfo transmissions which are **Not** send to NodeNum 4294967295 (this is not only Node-DB-overflow but I guess to a high percentage)
[image]

On the right side you can see the median time between the transmissions (I removed the node name on the left side).

My point is, even if a user uses the default value of 3h or longer, the node does not stick to it. 
I can understand that this is for various reasons not the case but the worst case minimum interval of either 1 or 5 minutes as per the current Firmware (in NodeInfoModule.cpp) can generate high traffic (yes, as long as channel utilization is below 25%)

b8b8, 2025-03-22:
Good data. 

I think one of the main problems is you have roughly 40 routers.

My mesh size is currently 358 and we have 5 routers.

GUVWAF, 2025-03-22:
Nice data indeed. 
Newer firmware will not request NodeInfo from newly heard nodes if their NodeDB is full. I believe this accounts for most of the NodeInfos you're seeing, and thus it should automatically become better once people update their nodes.

MeshTrafficObserver, 2025-03-22:
Thank you all. 
Indeed! Now I found the condition "&& !nodeDB->isFull" in the respective part of the MeshService.cpp.

We will issue some calls to all our Swiss community to update their firmware.

closed 2025-03-22 by MeshTrafficObserver