meshtastic/firmware #6375: [Bug]: Meshtastic 2.6.2: Special characters are not displayed on the device

az5456, 2025-03-23:
### Category

Other

### Hardware

Heltec Vision Master E290

### Firmware Version

2.6.2.31c0e8f

### Description

Special characters are not displayed on the device. There aren't even placeholders like □, the letters are simply ignored. This renders some words completely unreadable in non-english languages, e.g. "Schöne Grüße" is displayed as "Schne Gre".

### Relevant log output

```Shell

```

caveman99, 2025-03-23:
the stock firmware supports umlauts. Is this running the inkHUD?

az5456, 2025-03-23:
Yes, it is running inkHUD.

caveman99, 2025-03-23:
@todd-herbert can you have a look here? do you use the same character sets we use for the Oled UI or is this a different lib?

todd-herbert, 2025-03-23:
Right now, InkHUD only has the basic ASCII character set, but the mechanism is there to extend it. There's a proof of concept for Cyrillic fonts (Windows-1251 encoding). My intention is to also offer Windows-1250 and Windows-1252 encoding. 

Because the encodings overlap, selecting one will mean building a custom firmware, although Windows-1252 (with umlauts) may very well become the default, if we have the flash space for the extended fonts.

As a quick alternative right now, before the custom fonts are made, it's possible to make suitable ASCII substitutions in `nicheGraphics.h`: https://github.com/todd-herbert/meshtastic-firmware/tree/InkHUD/src/graphics/niche/InkHUD/docs#utf-8-substitutions

caveman99, 2025-03-23:
thanks for the explanation, i would keep the charsets aligned with the OLED stuff. So having cyrillic or greek characters requires a recompile with the OLED flag. Standard would be western ascii (1252) codepage.

rekab72, 2025-03-24:
Would this also be the cause of not seeing emojis (waving hand) sent from a device on 2.5.23 to T-Deck+ on 2.6.2?

Mhowser, 2025-05-11:
Basic block characters are not visible on the oled screen for my LILYGO T-Beam, will #6768 fix this?

todd-herbert, 2025-05-12:
@Mhowser No, the OLED displays have their own separate graphics code. Which characters are not visible though?

Mhowser, 2025-05-13:
These characters do not render on the OLED screen: https://en.wikipedia.org/wiki/Block_Elements I would like these for displaying scannable QR codes.

todd-herbert, 2025-05-13:
> These characters do not render on the OLED screen: https://en.wikipedia.org/wiki/Block_Elements I would like these for displaying scannable QR codes.

Oh, no sorry, those chatacters definitely aren't supported, but I don't believe the T-Beam's display has enough resolution to display a QR code anyway.

closed 2025-05-22 by thebentern