meshtastic/firmware #6380: [Feature Request]: External notification over MQTT

Akstef, 2025-03-23:
### Platform

ESP32

### Description

Hello

I have already opened an issue for the same request but it was closed without giving me a solution
https://github.com/meshtastic/firmware/issues/6138

So again, I am able to send a text message over MQTT but if I send the bell character (alt + 7) the receiver is not triggering the external notification and only display •
Sending the bell character from the android app (using the bell icon shortcut) is working fine

Request: Trigger the external notification over MQTT

garthvh, 2025-03-23:
The character is included in the text of the alert message, if you are sending it manually and it fails something is wrong with your encoding.

Akstef, 2025-03-24:
I use home assistant to send the message so I have activated json. When I use alt + 7 it display ● in the message and I recieve it but no external notification is triggered. Only ● is displayed.

I have tried with mqtt explorer (also in json format) and it also only diplay ● without triggering the notification.

Maybe it is not possible to trigger the notification when using json but how to send the bell character without json? What will be the encoding?

madeofstown, 2025-03-24:
If you copy the bell emoji from some text where it shows up properly and paste it into the text field in Home Assistant does it work as expected?

garthvh, 2025-03-24:
> If you copy the bell emoji from some text where it shows up properly and paste it into the text field in Home Assistant does it work as expected?
> 
> Pretty sure that emoji are Unicode.

Emoji are not the same as the alert bell character, and not supported by the json mode.

Akstef, 2025-03-24:
> > If you copy the bell emoji from some text where it shows up properly and paste it into the text field in Home Assistant does it work as expected?
> > Pretty sure that emoji are Unicode.
> 
> Emoji are not the same as the alert bell character, and not supported by the json mode.

Yes, I confirm that I cannot send any Emoji in json. Is there any possibility to send the bell character in json? What needs to be done to send it in MQTT without json activated?

garthvh, 2025-03-24:
It is already sent via the protobufs over mqtt like any other character, not sure there is any home assistant utility for the bell character, which is what the json mode is for.

Akstef, 2025-03-24:
Ah! So if I understand well it is not possible to sent the bell character from home assistant.

Do you know if it is possible to send it from mqtt explorer?

closed 2025-05-17 by github-actions[bot]