meshtastic/firmware #6403: [Feature Request]: UDP unicast instead of multicast

formtapez, 2025-03-27:
### Platform

Cross-Platform

### Description

It would be great if the user could configure a custom UDP unicast address (a "normal" IP-address, or even better: a complete hostname which then gets resolved) for the UDP-Link feature. That would allow to bridge two nodes which are not located in the same WiFi network.
Then other infrastructure (like the HAMNET for example) can be used to bridge two networks without having to put everything in a L2-VPN.

Jorropo, 2025-03-27:
👍 notes from discord we should encrypt theses, DTLS probably makes sense here.

Also ever since v2.6.2 I've made two changes (96ba94843b25f787148e3353a7476ed175518b3b & f41afb14b15aed9f565df93f341a5bff3120f11b) which means a running a single L2 domain is no longer required.
You can setup an L3 IGMP based routing solution / VPN and it should work, see https://an0n-r0.medium.com/making-dlna-through-site-to-site-vpn-work-f393629f4ce0 for example to set that up with wireguard (an L3 VPN).

formtapez, 2025-03-27:
Okay, but therefore you need two additionaly computers which cost money and energy just to manage that VPN and make the multicast-routing. The cheapest solution would be just to be able to change that IP address. Then a normal port-forward could be used, or just use existing routing (HAMNET).
I dont think that additionaly encryption is needed, because the UDP-Link is just a replacement for the air-interface, which is already encrypted.