meshtastic/firmware #6423: Boring node eviction logic evicts interesting nodes, is mean to amateur radio operators

rfschmid, 2025-03-27:
Change [Add logic to nodeDB to prefer evicting boring nodes](https://github.com/meshtastic/firmware/pull/4441) causes undesired behavior for me. 

We have a large amateur radio community near my home/work who are also operating meshtastic nodes. This means dozens of nodes I see every day have the "Licensed Amateur Radio" flag set, and thus encryption disabled.

Because of change #4441, these nodes that I'm in range of all day every day and I am interested in are evicted from my nodeDB multiple times per day, while nodes that I only saw while traveling months ago, who I never interacted with, who are hundreds of miles away, and who I will likely never see again are kept around instead because they're "interesting" just by virtue of not being hams I guess.

As a result I get new node seen notifications on my phone pretty much constantly (many, many times per day) for the same group of nodes, that keep trading places in my nodeDB. If we evicted old nodes instead, I'd have enough space for the ones I see often and would only get new node notifications for nodes that are actually likely to be new to me. 

It's rude to treat people as boring just because they're hams. They're only boring if you get them going about radios. Free the hams. Thanks.

garthvh, 2025-03-27:
Nodes with and without encryption enabled should not be interacting, sounds like you are seeing a bunch of improperly set up ham mode nodes, if you update your firmware using empty keys without ham mode has been removed and when in licensed mode you would have a blended mesh anymore.

closed 2025-03-27 by garthvh

rfschmid, 2025-03-27:
I guess it's possible that these guys all just have out of date nodes and put their callsign as their name just for fun, but either way the boring node logic here is still resulting in really annoying behavior :/

rfschmid, 2025-03-28:
Could we consider modifying the "boring node" logic to either not replace one "boring" node with another "boring" node, but instead drop the oldest node, or, if nodes with and without encryption shouldn't be interacting, not bump a node from the nodeDB if we're going to replace it with an unencrypted node?

garthvh, 2025-03-28:
> Could we consider modifying the "boring node" logic to either not replace one "boring" node with another "boring" node, but instead drop the oldest node, or, if nodes with and without encryption [...]

Time is one of the elements used, but this mixed hams and encrypted users behavior you are seeing is a mis configuration that is fixed in newer firmware versions 

https://github.com/meshtastic/firmware/issues/5700

The existence of a call sign in the long name does not mean ham mode is actually enabled.  Nodes with PKC keys are kept in the node db over a node without to reduce key mismatch errors.