meshtastic/firmware #6430: [Bug]: name lost

cracky22, 2025-03-28:
### Category

Other

### Hardware

Heltec V3

### Is this bug report about any UI component firmware like InkHUD or Meshtatic UI (MUI)?

- [ ] Meshtastic UI aka MUI colorTFT
- [ ] InkHUD ePaper
- [ ] OLED slide UI on any display

### Firmware Version

2.6.2

### Description

the heltec v3 keeps forgetting its name and you have to set it again

### Relevant log output

```Shell

```

benkyd, 2025-04-16:
I have been able to reproduce this - no steps to trigger it to happen. Please update if you can figure out exactly what causes it

cracky22, 2025-04-16:
@benkyd for me, this mostly occurs when the battery of my node runs out or after updates, but only after restarting after an update (shutdown + restart up)

jwheeler188, 2025-05-12:
> it is interesting that the problem has existed since 2.4.x and still exists in 2.6.x...

I may load the latest beta on my heltec and see if it still happens.