meshtastic/firmware #6431: [Feature Request]: Node DB Caching expansion on devices with SD Card or connected clients.

mattdrum, 2025-03-28:
### Platform

Cross-Platform

### Description

It has been frustrating in areas with a high node density that nodes will disappear from the list when the Node DB limit size has been exceeded.

Since there is a limitation for the Node DB size on the device memory, I would like to consider caching the Node info on devices with an SD or external storage. By doing so, the current limit may be exceeded with the external storage and more nodes can be recalled.

This could also include caching the nodes on Android, Apple, and other clients to keep a list of all nodes that have been seen.

garthvh, 2025-03-28:
iOS already does this and it will be in the next Android releaee

mikecarper, 2025-04-01:
why not use the ram size as defined here to figure out the max nodes?
https://github.com/search?q=repo%3Ameshtastic%2Ffirmware+%22maximum_ram_size%22%3A&type=code&p=1
vs the flash size as is done now
https://github.com/meshtastic/firmware/blob/master/src/mesh/mesh-pb-constants.h#L33

shalberd, 2025-04-21:
> iOS already does this and it will be in the next Android release

@garthvh since which IOS App release has this been the case?
And from which Android App release will this be the case?
Do you have the merged pull request number for IOS App Repo and merge request number for the Android App?

garthvh, 2025-04-21:
2.0, has always been part of the iOS app, should be 2.5.22 on Android.

shalberd, 2025-04-21:
@garthvh thank you. Regarding the node DB size limit of 100 on the devices themselves, on the radios / file system, I wonder how much engineering goes into increasing the node DB limit to more than 100. We have a lot of Citizens Band users and amateur radio users here trying Meshtastic, so I can easily see the number of active nodes going above 100.
What is the deal then with e.g. RAK 4630 hardware? Is any current hardware ready for a larger on-radio node DB?

garthvh, 2025-04-22:
> [@garthvh](https://github.com/garthvh) thank you. Regarding the node DB size limit of 100 on the devices themselves, on the radios / file system, I wonder how much engineering goes into increasing the node DB limit to more than 100. [@thebentern](https://github.com/thebentern) We have a lot of Citizens Band users and amateur radio users here trying Meshtastic, so I can easily see the number of active nodes going above 100 in the wider area. Wider area means more than 1 or 2 hops away, though, at least for now :-) What is the deal then with e.g. RAK 4630 hardware? Is any current hardware ready for a larger on-radio node DB? Is it even necessary for routing decisions and so on?

In a busy mesh favorite the nodes you want to dm

shalberd, 2025-04-22:
@garthvh yes, correct, having the ability to tag nodes as favorites, which would keep them on top of the list in the Android and IOS App GUI, but which would also keep them from being deleted from the 100 nodes on-radio node DB for as long as possible, correct. 
That might be a better way to go than to increase the on-radio node DB, which, like with RAK 4631 and its 256KB RAM and 1MB onboard flash, is not all that easy.

shalberd, 2025-04-22:
Android Google Play Store version is still very low and does not have e.g. node favoriting. For those who do not want to use sideloading or F-Droid, but Play Store :-)

closed 2025-04-22 by thebentern

b8b8, 2025-04-28:
> Android Google Play Store version is still very low at 2.5.16 and does not have e.g. node favoriting. For those who do not want to use sideloading or F-Droid, but Play Store :-) How do you decide when a version of the Android App is ready for Play Store?

Install obtanium app and point it to the GitHub directory for meshtastic. You can choose to only install the official Google play release and not have to wait for Google to finally get around to adding it.