meshtastic/firmware #6448: [Feature Request]: Disable Network Module on all NRF boards except for the Rak 4631

garthvh, 2025-03-29:
### Platform

NRF52

### Description

Network functionality is only available for the RAK 4631 NRF board

ponzano, 2025-03-29:
Would this remove the possibility to add an Ethernet module to other nrf boards?

garthvh, 2025-03-29:
> Would this remove the possibility to add an Ethernet module to other nrf boards?

There are no other nrf boards with this hardware available currently.

ponzano, 2025-03-29:
> > Would this remove the possibility to add an Ethernet module to other nrf boards?
> 
> There are no other nrf boards with this hardware available currently.

Good to know, but would this remove the possibility to add an Ethernet module to other nrf boards?
We are working on the MeshLink board and would like to keep this functionality 
Would we need to add an exception for us?

garthvh, 2025-03-29:
And the point that you make it work you can update the mesh link variant.

PR #6462 (2025-03-31): Disable network config for non-eth_gateway nrf52 and non-W RP2040 targets
Closes #6448
closed 2025-03-31 by thebentern