meshtastic/firmware #6453: [Feature Request]: Add channel selector to serial module TEXTMSG mode

linker3000, 2025-03-29:
### Platform

Cross-Platform

### Description

At the moment, the serial module's TEXTMSG mode sends its message over the default channel.

It would be very useful to have an optional channel selector message prefix to be able to select the channel over which the message should be sent,

 Example, to send on channel 1: "/1Hello World"

garthvh, 2025-03-29:
You want to use the protobuf mode for this.

closed 2025-03-29 by garthvh

linker3000, 2025-03-30:
This is a very disappointing reply because it dismisses the suggestion straight away without further chance for discussion by proposing an extremely complex to implement solution for simple sensors that can be set to output just a serial string value, and would require an intermediate device to handle the complexity of protobufs adding to the cost of implementing a simple remote sensor or info reporting device.