meshtastic/firmware #6460: [Bug]: RAK MQTT Ethernet Gateway Missing Time On Packets

Xaositek, 2025-03-30:
### Category

Other

### Hardware

Rak4631

### Is this bug report about any UI component firmware like InkHUD or Meshtatic UI (MUI)?

- [ ] Meshtastic UI aka MUI colorTFT
- [ ] InkHUD ePaper
- [ ] OLED slide UI on any display

### Firmware Version

2.6.3.640e731

### Description

I have observed that my RAK MQTT Ethernet Gateway is not sending Time with it's Device or Environmental Metrics Packets.

The below serial log is only to show that the device has actual time. I have NTP configured and also used `--set-time` to ensure it has the needed time.

```
% meshtastic --host 192.168.96.47 --get network    
Connected to radio
network.ntp_server: meshtastic.pool.ntp.org
network.eth_enabled: True
Completed getting preferences
```

[image]
[image]

### Relevant log output

```Shell
DEBUG | 11:34:58 40016 [mqtt] Use channel 0 (hash 0x8)
DEBUG | 11:34:58 40016 [mqtt] Expand short PSK #1
DEBUG | 11:34:58 40016 [mqtt] Use AES128 key!
DEBUG | 11:34:58 40017 [mqtt] decoded message (id=0x42e90ffb fr=0x537e8958 to=0xffffffff, WantAck=0, HopLim=7 Ch=0x0 Portnum=4 via MQTT hopStart=7)
DEBUG | 11:34:58 40017 [Router] Found existing packet record for fr=0x537e8958,to=0xffffffff,id=0x42e90ffb
DEBUG | 11:34:58 40017 [Router] Add packet record fr=0x537e8958, id=0x42e90ffb
DEBUG | 11:34:58 40017 [Router] Ignore dupe incoming msg (id=0x42e90ffb fr=0x537e8958 to=0xffffffff, WantAck=0, HopLim=7 Ch=0x0 Portnum=4 via MQTT hopStart=7)
DEBUG | 11:34:58 40017 [Router] Repeated reliable tx
DEBUG | 11:34:58 40017 [Router] Incoming msg was filtered from 0x537e8958
INFO  | 11:34:58 40017 [mqtt] Ignore downlink message we originally sent
DEBUG | 11:35:00 40019 [RadioIf] rx_snr found. hop_limit:6 rx_snr:6.000000
DEBUG | 11:35:00 40019 [RadioIf] rx_snr found in packet. Setting tx delay:2520
DEBUG | 11:35:03 40021 [RadioIf] Started Tx (id=0x42e90ffb fr=0x537e8958 to=0xffffffff, WantAck=0, HopLim=6 Ch=0x8 encrypted len=109 rxtime=2086018498 rxSNR=6 rxRSSI=-56 hopStart=7 relay=0x58
DEBUG | 11:35:03 40021 [RadioIf] Packet TX: 1050ms
DEBUG | 11:35:03 40021 [RadioIf] 0 packets remain in the TX queue
DEBUG | 11:35:04 40022 [RadioIf] Completed sending (id=0x42e90ffb fr=0x537e8958 to=0xffffffff, WantAck=0, HopLim=6 Ch=0x8 encrypted len=109 rxtime=2086018498 rxSNR=6 rxRSSI=-56 hopStart=7 rel
DEBUG | 11:35:07 40025 [RadioIf] Lora RX (id=0x42e90ffb fr=0x537e8958 to=0xffffffff, WantAck=0, HopLim=6 Ch=0x8 encrypted len=109 rxSNR=3 rxRSSI=-107 hopStart=7)
DEBUG | 11:35:07 40025 [RadioIf] Packet RX: 1050ms
DEBUG | 11:35:07 40025 [Router] Found existing packet record for fr=0x537e8958,to=0xffffffff,id=0x42e90ffb
DEBUG | 11:35:07 40025 [Router] Add packet record fr=0x537e8958, id=0x42e90ffb
DEBUG | 11:35:07 40025 [Router] Ignore dupe incoming msg (id=0x42e90ffb fr=0x537e8958 to=0xffffffff, WantAck=0, HopLim=6 Ch=0x8 encrypted len=109 rxSNR=3 rxRSSI=-107 hopStart=7)
DEBUG | 11:35:07 40025 [Router] cancelSending id=0x42e90ffb, removed=0
DEBUG | 11:35:07 40025 [Router] Incoming msg was filtered from 0x537e8958
DEBUG | 11:35:17 40036 [Power] Battery: usbPower=1, isCharging=1, batMv=4299, batPct=100
```

garthvh, 2025-03-31:
Can you run this test without doing the --settime which is not going to be a trusted time?

Xaositek, 2025-03-31:
Previous observation after filing this bug, timestamps starting showing up and worked throughout the day today.

I was able to get it from initial boot, observation is that despite the radio booting up with the correct IP, Gateway, and IP - the first NTP and initial MQTT connection is failing. Subsequent connection to the MQTT server succeeds. Even after 5 minutes though the NTP connection failed.

Connected to radio
INFO  | ??:??:?? 14 Local IP 192.168.96.47
INFO  | ??:??:?? 14 Subnet Mask 255.255.255.0
INFO  | ??:??:?? 14 Gateway IP 192.168.96.1
INFO  | ??:??:?? 14 DNS Server IP 192.168.96.1
DEBUG | ??:??:?? 14 LoRA bitrate = 118.394310 bytes / sec
INFO  | ??:??:?? 14 PowerFSM init, USB power=1
DEBUG | ??:??:?? 14 State: BOOT
DEBUG | ??:??:?? 14 [Power] Battery: usbPower=1, isCharging=1, batMv=4271, batPct=100
INFO  | ??:??:?? 14 [RangeTest] Range Test Module - Disabled
INFO  | ??:??:?? 14 [mqtt] Connecting directly to MQTT server mo-mesh.com, port: 1883, username: meshtastic, password: XXXXXXXX
**WARN  | ??:??:?? 14 [mqtt] Failed to connect to MQTT server**
INFO  | ??:??:?? 14 [ethConnect] Start Ethernet network services
INFO  | ??:??:?? 14 [ethConnect] Start NTP time client
INFO  | ??:??:?? 14 [ethConnect] API server listening on TCP port 4403
**INFO  | ??:??:?? 14 [ethConnect] Update NTP time from meshtastic.pool.ntp.org
ERROR | ??:??:?? 15 [ethConnect] NTP Update failed**
INFO  | ??:??:?? 15 [EnvironmentTelemetry] Environment Telemetry: init
INFO  | ??:??:?? 15 [EnvironmentTelemetry] No BME680 state found (File: /prefs/bsec.dat)
INFO  | ??:??:?? 15 [EnvironmentTelemetry] Init sensor: BME680 with the BSEC Library version 2.5.0.2 
INFO  | ??:??:?? 15 [EnvironmentTelemetry] Opened BME680 sensor on i2c bus
INFO  | ??:??:?? 17 [PowerFSM] Init the Bluefruit nRF52 module
INFO  | ??:??:?? 17 [PowerFSM] Bluetooth pin set to '123456'
INFO  | ??:??:?? 17 [PowerFSM] Init the Device Information Service
INFO  | ??:??:?? 17 [PowerFSM] Init the Battery Service
INFO  | ??:??:?? 17 [PowerFSM] Init the Mesh bluetooth service
INFO  | ??:??:?? 17 [PowerFSM] Set up the advertising payload(s)
INFO  | ??:??:?? 17 [PowerFSM] Advertise
INFO  | ??:??:?? 17 [PowerFSM] Disable NRF52 Bluetooth. (Workaround: tx power min, advertise stopped)

.......

**INFO  | ??:??:?? 90 [mqtt] Connecting directly to MQTT server mo-mesh.com, port: 1883, username: meshtastic, password: XXXXXXXX
INFO  | ??:??:?? 90 [mqtt] MQTT connected**
INFO  | ??:??:?? 90 [mqtt] Subscribe to msh/MO/2/e/LongFast/+
INFO  | ??:??:?? 90 [mqtt] Subscribe to msh/MO/2/e/PiperFamily/+
INFO  | ??:??:?? 90 [mqtt] Subscribe to msh/MO/2/e/Riding/+
INFO  | ??:??:?? 90 [mqtt] Subscribe to msh/MO/2/e/CMRG/+
INFO  | ??:??:?? 90 [mqtt] Subscribe to msh/MO/2/e/Skywarn-F/+
INFO  | ??:??:?? 90 [mqtt] Subscribe to msh/MO/2/e/PKI/+
DEBUG | ??:??:?? 90 [mqtt] Publish enqueued MQTT message
INFO  | ??:??:?? 90 [mqtt] publish msh/MO/2/e/LongFast/!6fae2558, 143 bytes from queue

........

INFO  | ??:??:?? 316 [ethConnect] Update NTP time from meshtastic.pool.ntp.org
ERROR | ??:??:?? 317 [ethConnect] NTP Update failed

**Testing the NTP Server, seems like intermittently the host isn't responding and I'm getting unlucky**
% ping meshtastic.pool.ntp.org
PING meshtastic.pool.ntp.org (23.155.40.38): 56 data bytes
Request timeout for icmp_seq 0
Request timeout for icmp_seq 1
^C
--- meshtastic.pool.ntp.org ping statistics ---
3 packets transmitted, 0 packets received, 100.0% packet loss

% ping meshtastic.pool.ntp.org
PING meshtastic.pool.ntp.org (141.11.228.173): 56 data bytes
64 bytes from 141.11.228.173: icmp_seq=0 ttl=52 time=48.194 ms
64 bytes from 141.11.228.173: icmp_seq=1 ttl=52 time=42.262 ms
64 bytes from 141.11.228.173: icmp_seq=2 ttl=52 time=42.165 ms

Xaositek, 2025-03-31:
I kept monitoring logs and noticed this also.

We finally got the time and yet the next check, failed again.

INFO  | 20:20:51 617 [ethConnect] Update NTP time from meshtastic.pool.ntp.org
ERROR | 20:20:52 619 [ethConnect] NTP Update failed

Xaositek, 2025-03-31:
Monitored this for awhile - tested changing MQTT server from hostname to IP and back.

One observation - when NTP failed, the MQTT server forced a reconnect.

INFO  | 20:30:58 1225 [ethConnect] Update NTP time from meshtastic.pool.ntp.org
ERROR | 20:31:00 1226 [ethConnect] NTP Update failed
WARN  | 20:31:00 1226 [EnvironmentTelemetry] runTrigger BSEC2 code: 100
INFO  | 20:31:00 1226 [mqtt] Connecting directly to MQTT server mo-mesh.com, port: 1883, username: meshtastic, password: XXXXXXX
INFO  | 20:31:00 1226 [mqtt] MQTT connected
INFO  | 20:31:00 1226 [mqtt] Subscribe to msh/MO/2/e/LongFast/+
INFO  | 20:31:00 1226 [mqtt] Subscribe to msh/MO/2/e/PiperFamily/+
INFO  | 20:31:00 1226 [mqtt] Subscribe to msh/MO/2/e/Riding/+
INFO  | 20:31:00 1226 [mqtt] Subscribe to msh/MO/2/e/CMRG/+
INFO  | 20:31:00 1226 [mqtt] Subscribe to msh/MO/2/e/Skywarn-F/+
INFO  | 20:31:00 1226 [mqtt] Subscribe to msh/MO/2/e/PKI/+

Will upgrade to Meshtastic 2.6.4 and monitor

Xaositek, 2025-04-08:
Closing this for now - reproduction is not reliable and I'm tempted to think it's related to a cellular connection lso.

closed 2025-04-08 by Xaositek