meshtastic/firmware #6467: Possible unintended discrepancy in rate limiting of Adhoc Ping

todd-herbert, 2025-03-30:
### Discussed in https://github.com/meshtastic/firmware/discussions/6442

<div type='discussions-op-text'>

<sup>Originally posted by **todd-herbert** March 29, 2025</sup>
In the OLED UI, double pressing the user button sends an "adhoc ping". This is performed by [`MeshService::trySendPosition`](https://github.com/meshtastic/firmware/blob/0491c890d72981c360ddbfebbf314060f031c43b/src/mesh/MeshService.cpp#L259-L280).

If the node has a valid position, this ping takes the form of a position packet. If position is not known, the ping takes the form of a node info packet. It looks to me that, if the ping takes the form of a position packet, it can be sent as often as the user desires, but if it takes the form of a node info packet, it is subject to the usual five minute cool-down for node info broadcasts. 

Is this rate-limiting discrepancy intentional?
Is there a desire to adjust the rate-limiting in some way to align the two cases?
Currently there is no feedback to a user when a ping is rejected, other than in the debug logs. Should the cool-down be indicated in some way? </div>

garthvh, 2025-03-30:
It is not an accident that the position packet is not rate limited, position is what the feature actually is intended for and portion packets are both smaller and potentially more important, while node info packet is larger and less useful for safety.  Rate limiting both has not been necessary to date.

todd-herbert, 2025-03-30:
Oh I totally accept the difference in the need for rate limiting. I'm just thinking specifically about their implementation as the "adhoc ping" feature. The result there is that the "ping" is sometimes not sent, with no indication of this to the user.

todd-herbert, 2025-03-30:
What I'm really looking for is feedback on how to handle this discrepancy. Should we standardize the cool-down time between adhoc pings? Should the user be informed when the cool-down period is in effect? Should adhoc ping be split and exposed explicitly to the user as separate adhoc position and adhoc node info features?

garthvh, 2025-03-30:
Client notifications is the right way to handle this in that case.

garthvh, 2025-03-30:
And then InkHUD could display those notifications like the apps do.

todd-herbert, 2025-04-03:
Oh neat, I hadn't actually come across the client notification feature so far. I could see the value of having implementing that within InkHUD in the future.

Would you definitely like a connected client app to be notified if an ad-hoc ping fails to send? For all devices, or only those which do not have a display?

garthvh, 2025-04-03:
It is for all devices

closed 2025-05-26 by github-actions[bot]