meshtastic/firmware #6478: [Board]: Seeed Xiao ESP32S3 + SX1262 for xiao (not the kit)

puflet, 2025-04-01:
### SOC

ESP32

### Lora IC

SX1262

### Product Link

https://www.seeedstudio.com/Wio-SX1262-for-XIAO-p-6379.html

### Description

Hi Meshers!
I have bought the board from the link, as I had some "Xiao ESP32-S3" laying around and decided I do not need the kit.
It turned out, that the board in question uses different pins for the LoRA chip, as it does not use Seeed's 'sense' connector, but old-school soldiered pins. I have managed to rewrite the firmware by only making changes to pins_arduino.h and variant.h files.
I think it is useless to add a new HardwareModel, but maybe a board variant, with proper explanation of the difference between the sold "Kit" version and the alternative "DIY" approach.

madeofstown, 2025-04-02:
I'm very interested in your changes. I am working on a carrier board for Xiao and a 1 watt radio.Alreasy have the nRF version working but it would save me some time if I could look at your variant files for the ESP32S3 version.

puflet, 2025-04-03:
Nothing spectacular, only rewritten some of the pins.
The boards I have used:
[SKU: 113010003 Wio-SX1262 for XIAO](https://www.seeedstudio.com/Wio-SX1262-for-XIAO-p-6379.html)
[SKU: 113991114 Seeed Studio XIAO ESP32S3](https://www.seeedstudio.com/XIAO-ESP32S3-p-5627.html)
**pins_arduino.h**
`static const uint8_t SS = 5; // was 41
`
**variant.h**
```
#define HAS_GPS 0 // (not actually required, on the test board I do not plan using GPS)
//#define USCREEN_SSD1306 // (as above)

#define LORA_CS 5 // was 41
#define LORA_RESET 3 // was 42
#define LORA_DIO1 2 // was 39

//#define LORA_DIO2 5 // was 38 (not sure about that, could not find DIO2 exposed according to Seeed's documentation)
#define SX126X_BUSY 4 // was 40
#define SX126X_RXEN 6 // was 38
```

Ope8888, 2025-05-18:
> Rien de spectaculaire, j'ai juste réécrit certaines broches. Les cartes que j'ai utilisées : [SKU : 113010003 Wio-SX1262 pour XIAO ](https://www.seeedstudio.com/Wio-SX1262-for-XIAO-p-6379.html) [SKU : 113991114 Seeed Studio XIAO ESP32S3](https://www.seeedstudio.com/XIAO-ESP32S3-p-5627.html) J'ai utilisé /variants/seeed_xiao_s3/variant.h comme modèle et effectué les modifications décrites ci-dessous.
> 
> **pins_arduino.h**
> 
> ```
> static const uint8_t SS = 5; // was 41
> ```
> 
> **variante.h**
> 
> ```
> #define HAS_GPS 0 // (not actually required, on the test board I do not plan using GPS)
> //#define USCREEN_SSD1306 // (as above)
> 
> #define LORA_CS 5 // was 41
> #define LORA_RESET 3 // was 42
> #define LORA_DIO1 2 // was 39
> 
> //#define LORA_DIO2 5 // was 38 (not sure about that, could not find DIO2 exposed according to Seeed's documentation)
> #define SX126X_BUSY 4 // was 40
> #define SX126X_RXEN 6 // was 38
> ```

Hello, I also bought this module separately!
I'm not comfortable with firmware!

Could someone provide a modified firmware ready to flash?

Thanks :)

Sabering1, 2025-07-29:
I feel positive since its enabling more devices to be used or old dead ones to be fixed, i would just add a note mentioning that I2C is not enabled for that device in that case

ab0oo, 2025-07-29:
It might be possible to use UART TX as an I2C pin, since even with a GPS attached, there's no real reason to send data TO a GPS.  I'll see if I can get baseline stuff working, then move towards I2C.

closed 2025-09-26 by github-actions[bot]