meshtastic/firmware #6488: [Bug]: Can't use Pin 0 for GPS

macvenez, 2025-04-03:
### Category

Other

### Hardware

DIY

### Is this bug report about any UI component firmware like InkHUD or Meshtatic UI (MUI)?

- [ ] Meshtastic UI aka MUI colorTFT
- [ ] InkHUD ePaper
- [ ] OLED slide UI on any display

### Firmware Version

2.5.18

### Description

Hi, we (me and @ponzano ) used PIN 0 (0.00) on our Meshlink (DIY) board for GPS enable pin but later we discovered it wasn't usable as setting value 0 for PIN_GPS_EN through firmware (or also apps) completely disables the usage of this pin instead of setting it.

We opened a pull request #6434 and we'd like to discuss and find a way to use this pin too.

As done for GNSS_AIROHA board we could add custom code for our board in order to force the usage of pin 0 but it may not be one of the cleanest solutions.

Another workaround we though of could be to use default value -1 (in firmware and from apps) to set a feature as not used so value 0 could also be used as real pin value but we don't know if there are any issues in doing this as in all fields value 0 is used and never -1.

Are there any updates on this? Does anyone have a possible solution?

Thanks

### Relevant log output

```Shell

```

garthvh, 2025-04-03:
Pin 0 should not be used.

ponzano, 2025-04-03:
> Pin 0 should not be used.

Why?

macvenez, 2025-04-03:
Why is that?
Can we at least add custom code as done for GNSS_AIROHA? Without this it wouldn't make any sense for us to even have the device added to Meshtastic firmware as one of the main functionality couldn't be used.

BTW on HT-CT62 which has very limited pins it would also be a great functionality to make it possible to use one of the few pins which are exposed

ponzano, 2025-04-03:
> Pin 0 should not be used.

Please elaborate, this isn't constructive as is

garthvh, 2025-04-03:
> > Pin 0 should not be used.
> 
> Please elaborate, this isn't constructive as is

You got an answer on discord and on the issue already.

macvenez, 2025-04-03:
> > > Pin 0 should not be used.
> > 
> > 
> > Please elaborate, this isn't constructive as is
> 
> You got an answer on discord and on the issue already.

The answer actually was "pin 0 is not used apart from button on esp32s3" which I might also be ok with...

I'd like to know if that's a protobuf or something else limitation and if it would be accepted to add a custom define similar to GNSS_AIROHA for our board.

We were suggested to avoid doing like GNSS_AIROHA but as it seems that's the only viable solution as "Pin 0 should not be used".

I'm ok with having answers contradicting my thoughts but I'd (at least) like to know why as we are contributing in developing Meshtastic too in a small way.

I think a project being open-source would also mean that anyone can and should be answered on why certain decisions were made in order to comprend what we are working on.

Thanks!

ponzano, 2025-04-03:
> > > Pin 0 should not be used.
> > 
> > Please elaborate, this isn't constructive as is
> 
> You got an answer on discord and on the issue already.

Well yes, on discord they said we can use the pin 0 as we need... just need to edit the firmware. Did you read something different? 
Btw it was also confirmed by Nordic (and the Datasheet) that pin 0.00 and 0.01 can be used as any other gpio if the board uses the internal nrf52 RC oscillator. 

[image]
[image]
[image]

closed 2025-05-26 by github-actions[bot]