meshtastic/firmware #6501: [Feature Request]: clock_format Protobuf Addition

Xaositek, 2025-04-06:
### Platform

NRF52, ESP32, RP2040, Linux Native, Cross-Platform

### Description

After review of the original "use_12h_clock", it became apparent that this isn't meeting the mark 100%.

Request:
- Deprecate / delete "use_12h_clock"
- Create "clock_format"
- Default value is "12"
- Other acceptable value is "24"

Xaositek, 2025-04-06:
CC: @HarukiToreda and @tropho23

garthvh, 2025-04-06:
Since there are only two values why does a book not work?

Xaositek, 2025-04-06:
> Since there are only two values why does a bool not work?

To get the desired effect; can we change "use_12h_clock" to a default enabled (true) state for any wipe/reload or new device?

Is there a one-time option to enable it during an upgrade?

tropho23, 2025-04-06:
I support making changes that benefit the majority of users; if we continue with the boolean setting we already have, is there a downside to making 12 hour format the default?  I imagine most users prefer this format but I might be wrong about that.  Is there a reason 24 hour format is the default, by virtue of this protobuf being disabled by default?

Also upgrading to firmware 2.6.x wipes the device so all settings are lost anyways...thus 12 hour clock format could be set as a default and please the (presumed) majority of users.

fifieldt, 2025-04-06:
Naw, 24 hour time is fairly common around the world. Europe, for example.

[image]

tropho23, 2025-04-06:
> Naw, 24 hour time is fairly common around the world. Europe, for example.
> 
> [image]

I don't disagree, and I prefer 24 hour clock myself.  I presume the majority of _Meshtastic_ users, not humans on the planet prefer 12 hour format.

fifieldt, 2025-04-06:
On that logic, probably need to also think about future meshtastic user-base too - where is growth likely? :) Bunch of US-based right now, for sure ... but in a few years time?

tropho23, 2025-04-06:
> On that logic, probably need to also think about future meshtastic user-base too - where is growth likely? :) Bunch of US-based right now, for sure ... but in a few years time?

No doubt, I agree 100%.  Choosing defaults is a no-win scenario but it makes sense to choose it for the largest user base.  I defer to @garthvh for stats on that since he can see some info for iOS users that may help.

HarukiToreda, 2025-04-06:
Just to add my two cents—in my experience, every cellphone I’ve bought has defaulted to the 12-hour format unless I manually change it. Not sure if that’s the case in other countries, especially in Europe, but that’s just what I’ve personally seen. It's more user friendly and a quick way to know those counting numbers are an actual clock.

garthvh, 2025-04-06:
US dominates is a large majority of use.  Only non English speaking country with much traffic is Germany which is about twice as large as the UK (US, Germany,UK, Canada). iOS manages this for users (and I think Android as well) which is probably why it has not come up a lot regarding oled screen.

tropho23, 2025-04-06:
> US dominates with a large majority of use. Only non English speaking country with much traffic is Germany which is about twice as large as the UK (US, Germany,UK, Canada). iOS manages this for users (and I think Android as well) which is probably why it has not come up a lot regarding oled screen.

Agreed, and given the OLED screens will now have a handy clock it's of new importance to device the fate of the default clock format for firmware 2.6. <insert dramatic music sting here>

jirogit, 2025-04-07:
It would be great to have an option like "Follow System Settings" that automatically adapts to the device's (phone, tablet, or OS) system time format preferences.

Is this something that could be considered as an additional feature?

Thanks for the awesome work!

thebentern, 2025-04-08:
After some back and forth discussion in discord, the conclusion was that we will stick to the current default 24 hour for 2.X (for now) because users do not like surprise default configuration changes. This will need to be configurable by the application display settings. 

Also a topic of conversation is, can we use lora region or tzdef to inform defaults in the future?