meshtastic/firmware #6518: [Bug]: Fan of the Particle Matter Sensor is rotating all the time

klikva, 2025-04-08:
### Category

Other

### Hardware

Rak4631

### Is this bug report about any UI component firmware like InkHUD or Meshtatic UI (MUI)?

- [ ] Meshtastic UI aka MUI colorTFT
- [ ] InkHUD ePaper
- [ ] OLED slide UI on any display

### Firmware Version

2.6.4.b89355f

### Description

I am using the Plantower PMSA003I (RAK12039) connected to the RAK19007/3 baseboard with the RAK4631. I've noticed that the fan runs continuously, even when it's not measuring. The rotation speed increases during measurements, but it continues to rotate slowly during the intervals between measurements. This behavior consumes a significant amount of energy when running on batteries; for instance, it uses 40% of a 4000 mAh LiPo battery overnight.

RAKwireless technical support mentioned that the fan can be powered down. Ideally, I would suggest the following cycle: the fan should operate at a low rotation speed for two minutes, followed by PM concentration measurement, and then turn off completely.

### Relevant log output

```Shell

```

vogon, 2025-04-11:
I'm building a device with a PMSA003I and noticed this same issue.  I've got a fix for it in https://github.com/meshtastic/firmware/commit/d2af4059d19e9dde748d94ec48680b1e3fe9146d, and would be happy to submit a pull request if that fix looks good to you and the maintainers.

klikva, 2025-04-22:
The fix is working great with the RAK4631 and PMSA003I. It’d be nice if the local reading interval (currently every minute) could be reduced to save more power.

PR #6569 (2025-04-29): PMSA003I: add support for driving SET pin low while not actively taking a telemetry reading
the Plantower PMSA003I sensor takes about 60-120mA of power at all times while the fan is running, as I discovered when I was building a solar-powered DIY weather station node.  now you can assign a GPIO in your variant file with `PMSA003I_ENABLE_PIN` which drives the SET pin on the sensor low to put it in standby mode.  tested on my DIY node with a fork of 2.5.20.4c97351.  

fixes #6518.

known todos:
- [x] verify that there are no regressions in `master`.
- [x] verify that it doesn't cause build failu [... cut]
closed 2025-04-29 by thebentern