meshtastic/firmware #6527: [Feature Request]: Time set automatically by defult via GPS or bluetooth conection

Techstuff313, 2025-04-08:
### Platform

ESP32

### Description

The lilygo T-watch S3 v2.6.4 has a bug that when the time zone is set the time is wrong. 
 
This can be fixed I am sure, but I am actually asking for a different way to do this. The T watch already needs to connect to a cellphone and already has a version with the GPS module. We should not need to manually set a time zone. It is still a watch. It would honestly be more user friendly if we had to manually adjust the watch than having to go online and finding the timezone code. 
 
This will allow the T-watch to be easier to operate for first time users who are not as involved in the technical aspect of the watch.

ALEXWM42, 2025-04-15:
Yes! The watch feature used to work. However, now it no longer has the correct time.

Xaositek, 2025-07-03:
Meshtastic 2.7 will now allow you to set Timezone on the TWatch itself, no phone required. As long as you get accurate time from GPS or another source, the Timezone can be set.