meshtastic/firmware #6529: [Bug]: fixes for device-install.sh and device-update.sh

mikecarper, 2025-04-09:
### Category

Other

### Hardware

Not Applicable

### Is this bug report about any UI component firmware like InkHUD or Meshtatic UI (MUI)?

- [ ] Meshtastic UI aka MUI colorTFT
- [ ] InkHUD ePaper
- [ ] OLED slide UI on any display

### Firmware Version

2.6.x

### Description

In device-install.sh add `ESPTOOL_CMD="$ESPTOOL_CMD --baud 1200"` like so
```
if "$PYTHON" -m esptool version >/dev/null 2>&1; then
	ESPTOOL_CMD="$PYTHON -m esptool"
elif command -v esptool >/dev/null 2>&1; then
	ESPTOOL_CMD="esptool"
elif command -v esptool.py >/dev/null 2>&1; then
	ESPTOOL_CMD="esptool.py"
else
	echo "Error: esptool not found"
	exit 1
fi
ESPTOOL_CMD="$ESPTOOL_CMD --baud 1200"

```

change the actual work being done lines from this
```
	echo "Trying to flash ${FILENAME}, but first erasing and writing system information"
	$ESPTOOL_CMD erase_flash
	$ESPTOOL_CMD write_flash 0x00 "${FILENAME}"
	echo "Trying to flash ${OTAFILE} at offset ${OTA_OFFSET}"
	$ESPTOOL_CMD write_flash $OTA_OFFSET "${OTAFILE}"
	echo "Trying to flash ${SPIFFSFILE}, at offset ${OFFSET}"
	$ESPTOOL_CMD write_flash $OFFSET "${SPIFFSFILE}"
```
to this
```
	echo "First erasing and writing system information"
	$ESPTOOL_CMD erase_flash
	sleep 5
	echo "Trying to flash ${FILENAME} at offset 0x00"
	$ESPTOOL_CMD write_flash 0x00 "${FILENAME}"
	sleep 5
	echo "Trying to flash ${OTAFILE} at offset ${OTA_OFFSET}"
	$ESPTOOL_CMD write_flash ${OTA_OFFSET} "${OTAFILE}"
	sleep 5
	echo "Trying to flash ${SPIFFSFILE}, at offset ${OFFSET}"
	$ESPTOOL_CMD write_flash ${OFFSET} "${SPIFFSFILE}"
```

In device-update.sh
replace
```
$ESPTOOL_CMD --baud 115200 write_flash 0x10000 ${FILENAME}
```
with
```
$ESPTOOL_CMD --baud 1200 write_flash 0x10000 ${FILENAME}
```

### Relevant log output

```Shell

```

caveman99, 2025-04-09:
why?

mikecarper, 2025-04-09:
Running it as is does not work. Forcing baud 1200 puts the device in firmware update mode. The sleep is needed as esptool can't find the device after writing; waiting 5 seconds makes it work.

closed 2025-06-01 by github-actions[bot]