meshtastic/firmware #6549: M5Stack CoreS3 - invalid header: 0xffffffff Error using web flasher and blank screen and no tx

masadams, 2025-04-09:
I posted this in another stale thread, but it could be considered a new/ongoing issue. I am brand-new to GitHub so apologies  in advance for missteps here. 

--------

> I realize this thread is a bit dated, but I encountered a similar issue when using the web flasher and wanted to share my experience in case it helps with ongoing development or support.
> 
> I consistently received the following error same as the OP:
> 
> invalid header: 0xffffffff
> 
> I tested the latest firmware builds (v2.5.20.4c97351, both beta and alpha) without success from the web flasher. 
> 
> Ultimately, I was able to manually flash the v2.5.20.4c97351 "firmware-m5stack-cores3-2.5.20.4c97351"  firmware via the console, which allowed the device to boot and partially function. However, several issues remained even after tying the alpha and nightly builds as well:
> 
> The display does not initialize (remains black).
> 
> I came across a comment  suggesting a potential SPI conflict between the LCD and LoRa on the CoreS3, which may be related, but I'm just guessing, complete amateur here.
> 
> The device is accessible in the Meshtastic web client and via Bluetooth, and it detects other nodes close by, but it is unable to send messages, and I am unable to see it from another device.
> 
> I’ve tried multiple build variants with the same results.
> 
> Hardware details:
> 
> M5Stack CoreS3
> 
> LoRa868 V1.1 module
> 
> There are dipswitch options on the LoRa Board, I used for NSS 6, for IRQ 14 and RST 5 which I took from the variants/m5stack_cores3/variant.h
> 
> I'm relatively new to the platform but happy to assist with testing or provide logs and configuration details if needed. I appreciate the work being done here and would be glad to contribute in any way I can to help resolve these issues. 

 _Originally posted by @masadams in [#3970](https://github.com/meshtastic/firmware/issues/3970#issuecomment-2788073146)_

tkiapril, 2025-05-17:
Unsure if I'm experiencing something similar, but my M5CoreS3 with the firmware only transmits when USB power is connected.

masadams, 2025-06-06:
Which firmware are you using? Does your display show anything on the M5CoreS3 ?

closed 2025-07-29 by github-actions[bot]