meshtastic/firmware #6557: BME280 on DIYV1 device

dboa, 2025-04-10:
I just build a new DIY meshtastic device using a ESP32-WROVER Dev Kit C V4
It's working fine and now I want to add a bme280 telemetry sensor.
As far as I could google, It goes in the pins 41 and 42 from most lora boards, but the ESP32 WROVER doesn't have pins 41 and 42.
Coudn't find the information on meshtastic/firmware/blob/master/variants/diy/v1/variant.h
Is telemetry sensors implemented on DIY boards?
What are the pins?

Thank you

Andrik45719, 2025-04-10:
You can redefine pins in the variant.h
```
#define I2C_SDA 21
#define I2C_SCL 22
```

dboa, 2025-04-10:
Those are the pins to the OLED display...
I can't have both?

madeofstown, 2025-04-10:
> Those are the pins to the OLED display... I can't have both?

Yes, i2c devices have an address.

Andrik45719, 2025-04-10:
> Those are the pins to the OLED display... I can't have both?
 I2C can handle multiple devices on one bus

dboa, 2025-04-10:
Yep it works, both OLED and Sensor are working in the same pins, this is so awesome.
Thank you all

closed 2025-04-10 by dboa